bastibl/gr-ieee802-11 #192: NameError: name 'ieee802_11' is not defined

harrihaven2, 2019-09-19:
Hello!

I am receiving this error when I try to open any of the examples. I have read [this related issue](https://github.com/bastibl/gr-ieee802-11/issues/6), and the issue is probably the python path given that I can't import it by running:
```
python
import ieee802_11
```
I am just having trouble discerning where the gr-ieee802-11 make is trying to install in python. Is there anyway I can manually add the ieee802_11 library into my python3.6 installation? Or what am I doing wrong?

Thank you for your patience and help

bastibl, 2019-09-19:
Looks like you forgot to paste the actual error message. (or what do you mean with "this error")

harrihaven2, 2019-09-19:
Apologies. 
The error I get from python is:
```
Python 3.6.8 (default, Aug 20 2019, 17:12:48) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'ieee802_11'
```

The error from gnu is:
```
>>> Loading: /home/ltmcmill/gr-ieee802-11/examples/wifi_phy_hier.grc
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block chan_est
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block encoding
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block chan_est
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block encoding
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block chan_est
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block encoding
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block chan_est
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block encoding
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
>>> Load Error: /home/ltmcmill/gr-ieee802-11/examples/wifi_phy_hier.grc: Flowgraph invalid
>>> Done
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/params/param.py", line 228, in evaluate
    value = self.parent_flowgraph.evaluate(expr)
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 249, in evaluate
    return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace))
[... 57 lines cut ...]
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/VariableEditor.py", line 197, in set_value
    evaluated = str(block.params[key].evaluate())
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/params/param.py", line 230, in evaluate
    raise Exception('Value "{}" cannot be evaluated:\n{}'.format(expr, e))
Exception: Value "[ieee802_11.LS, ieee802_11.LMS, ieee802_11.STA, ieee802_11.COMB]" cannot be evaluated:
name 'ieee802_11' is not defined
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/params/param.py", line 228, in evaluate
    value = self.parent_flowgraph.evaluate(expr)
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 249, in evaluate
    return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace))
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/VariableEditor.py", line 197, in set_value
    evaluated = str(block.params[key].evaluate())
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/params/param.py", line 230, in evaluate
    raise Exception('Value "{}" cannot be evaluated:\n{}'.format(expr, e))
Exception: Value "[ieee802_11.LS, ieee802_11.LMS, ieee802_11.STA, ieee802_11.COMB]" cannot be evaluated:
name 'ieee802_11' is not defined

```

bastibl, 2019-09-19:
OK, so it looks like you installed GNU Radio manually (i.e., not through the package manager). If GNU Radio works in general (i.e., you can run flowgraphs that come with GNU Radio), your PYTHONPATH etc, should be set up correctly. In that case, you probably didn't install the IEEE 802.11 module in the same location as GNU Radio (in your case `/usr/local`). Set the `CMAKE_INSTALL_PREFIX` to `/usr/local`

harrihaven2, 2019-09-19:
Ahhh! Thank you so much!! That worked! Although the gr-foo blocks are still unrecognized. Any idea how I can get some debug information on them? There are no errors in the console, but they are red in gnu-companion and have the "Missing Block" title.

bastibl, 2019-09-19:
You probably also installed the `gr-foo` OOT in the wrong directory.

closed 2019-09-20 by harrihaven2