bastibl/gr-ieee802-11 #194: Cannot connect USRP N210 to WLAN

Nico115, 2019-10-16:
greetings Bastian
First,I want to thank you for your work, it has been very useful for me.
I am using your contribution to develop my project but I cannot find the way to develop it 
at all.
I am trying to communicate an array of characters coming from a PC to USRP N210 using the 2.4GHz band with the purpose of upload those data to a web server using the esp32 module.
I have been working with your example of tranceiver. It works by sending a message strobe and
receiving data in 5Ghz channels, I checked it using the GNU radio console and WireShark but i don't know why it doesn't work in 2.4GHz channels.
I tried to use a gateway that could connect all devices but I can not recieve propertly none of those data packets checking WireShark.
Could you give me any advice to solve my problem please? I'll appreciate it a lot.

Nico115, 2019-10-22:
We are trying to send "Hello world" message to ESP32 which is connected to a WLAN network,I have the WiFi MAC block set up as: SRC MAC is the MAC of the USRP, and the DST MAC is the MAC of the ESP32. Setting it to 20MHz and 11g it does not send the information,I do not know how to configure the blocks on GNU Radio in order to connectthe USRP to the same NETWORK and thus be able to send data to the ESP32. The Host of the PDU socket, is the host that was configured in the ESP32 in UDP SERVER mode.
Could you tell me if I'm doing something wrong please? Thank you again.
[image]

Nico115, 2019-10-22:
I'm trying to send "Hello world" message to ESP32 which is connected to a WLAN network,I have the WiFi MAC block set up as: SRC MAC is the MAC of the USRP, and the DST MAC is the MAC of the ESP32. Setting it to 20MHz and 11g it does not send the information,I do not know how to configure the blocks on GNU Radio in order to connectthe USRP to the same NETWORK and thus be able to send data to the ESP32. The Host of the PDU socket, is the host that was configured in the ESP32 in UDP SERVER mode.
Could you tell me if I'm doing something wrong please? Thank you again.
[image]

bastibl, 2019-10-22:
I don't know why you changed the flow graph so heavily. Why don't you just use the unmodified tx flowgraph for a first simple test. Try to receive the frame with a normal WLAN card in monitor mode. If this works, switch to the ESP32 (I don't know if it also has monitor mode). If this works, there is at least physical connectivity.

There seems to be a large misunderstanding regarding networking. The socket PDU block and a UDP server on the ESP don't have anything to do with each other. The socket PDU would just allow you to inject the "hello world" message through UDP instead of using the message strobe block. However, it doesn't add any IP or UDP headers, which would be needed to receive a UDP packet on the ESP.

If you want to set up real networking, you have to configure a TUN/TAP device, ARP, and IP. See the nic.sh script in the apps folder. (You'll have to know a bit about networking to get that running). Then just send a UDP frame (for example with netcat or a simple Python script) outside of GNU Radio.

closed 2019-12-21 by bastibl