bastibl/gr-ieee802-11 #195: Rftap metadata origin

vChavezB, 2019-11-17:
I'm using your repo with gnuradio and a hackrf device. I haven't find information of how the Rftap metada is obtained.

Does the Information like antenna signal come independently of the location of where I put the SDR device or does it depend on the location of the SDR. I'm trying to measure some wifi packets signal strength depending on where I put the SDR device but don't know if this is possible with your work.

Thanks in advanced,
Victor

bastibl, 2019-11-19:
Do you maybe mean Radiotap instead of RFtap? The latter is not port of the default flow graphs. Radiotap is the format that is also used in WLAN monitor mode and implemented in the Wireshark connector block.
Since the SDR is no calibrated device, it's not (easily) possible to map average sample amplitude to an absolute power level. The only thing that's estimated is the SNR. The code sets noise to 0 and the antenna signal to the SNR, such the difference between signal and noise matches the SNR. Now that I'm writing it, it might make more sense to set the signal to dBFS and the noise to match the SNR...

Current code:
https://github.com/bastibl/gr-foo/blob/maint-3.8/lib/wireshark_connector_impl.cc#L105-L111

vChavezB, 2019-11-19:
> Do you maybe mean Radiotap instead of RFtap? The latter is not port of the default flow graphs. Radiotap is the format that is also used in WLAN monitor mode and implemented in the Wireshark [...]

I was asking since I tried different environments and was getting much or less the same signal strength value. As you suggest, I think that would be a good idea to have a better understanding of the value in the radiotap header.

closed 2019-12-21 by bastibl