bastibl/gr-ieee802-11 #196: phase tracking with pilot symbols

sayaz, 2019-11-19:
Hi,

1. In your implementation or as the standard says we are transmitting different pilot sequence for each symbol. Why we cannot use just a single pilot sequence for all the symbols. Say we select the pilot symbols to be [1,1,1,-1]. Why can't we transmit this over all the symbols? 

2. Can you share the formula that is used to estimate the phase offset from the pilot symbols?

bastibl, 2019-11-20:
1: It's a trade-off between throughput and robustness of the PHY. The standard decided for this particular trade-off. It only uses block pilots in the long training sequence.
2: No idea what phase offset you are talking about. But, in general: you expect a given bit, which maps to a BPSK constellation point. What you receive is a complex number with a different angle. The difference between the phases of the expected constellation point and the received one is an estimate for the phase offset.

sayaz, 2019-11-20:
Regarding question2. What I intended to ask is how we correct the phase rotation with pilot. Like for every symbol there are 4 pilot samples. Do we sum them up, and compare to the sum of original transmitted samples to find the angle? And then apply the difference to the OFDM symbol for correction ?

bastibl, 2019-11-24:
It depends on what effect you are trying to compensate. But for example: if the OFDM frame is not perfectly aligned in time domain to the transmitted frame, it introduces a linear phase shift, which can be estimated and compensated with the pilots in the way you are suggesting.

closed 2019-12-21 by bastibl