bastibl/gr-ieee802-11 #198: channel error in  wireshark and power of signal

adjmxx, 2019-11-22:
Hi, good night my name is Anderson Diaz, i'm studying in university of Carabobo Venezuela, i'm doing an networks analyzer WLAN with you receiver ieee802.11a/g/p is my theme of thesis, i want adjunct an spetrograme and sweep in frequency automatic; but i have a problem  with frames in wireshark, when see the frames in wireshark "802.11 Radio information"  the channel is 178MHz but will should be  2417MHz that is channel in the wich my router is transmittering.

Exists the possibility that can you help me. Thaks!!

adjmxx, 2019-11-22:
I'm using the  hackrf one

adjmxx, 2019-11-22:
[image]

bastibl, 2019-11-24:
This value is hard-coded at the moment. If it is crucial for you, you'd have to add the current frequency/channel as meta data to the frame. At the moment, only the SNR and the MCS is annotated. See the code for these two values to see how you can add more metadata.

bastibl, 2019-11-25:
you're welcome

closed 2019-11-25 by bastibl