bastibl/gr-ieee802-11 #199: Sending custom OFDM data symbols

namyh18, 2019-11-22:
Hi, @bastibl 

I'm trying to send same OFDM data symbols in each frame in "wifi_loopback.grc". I have some questions.

1. In your implementation, current_symbol[0] and [1] is long training field(LT) and [2] is Signal and rest of the symbols are Data symbols. I found that data symbols length depends on pdu_length which is just concatenation of "x" in Message strobe block.
When I set pdu_length to 0, it sends 11 symbols(except LT&Signal), and every +3 "x"s("xxx"), one symbol is added.
Then what is 11 symbols with empty message PMT(pdu_length == 0)?

2. Is giving series of "x"s equal to giving same OFDM data symbols?

3. How can I send data symbols which is equal to LT? Does LT equal to "sync_long_impl::LONG" or "base::LONG[]"?

[image]
* Sending same OFDM symbols mean, give same data symbol for Data_0 ... Data_K. And for Q.3, it means T_0 == T_1 == Data_0 ... Data_K.

bastibl, 2019-11-24:
1: It still adds a WLAN header and CRC. If you use BPSK 1/2, you have 24 data bits per symbol. So adding 3 'x's, i.e., 3 bytes adds one OFDM symbol.

2: sync_long::LONG is time domain (used for matched filtering), base::LONG is frequency domain (used to calculated csi)

3: Depends: in an ideal world it's not possible, since the WLAN card should use a pseudo-random scrambler seed for each frame,  i.e., you cannot know what your payload bits are mapped to. If you disable the scrambler, it's still complicated since you have to reverse interleaving, viterbi decoder, etc. There are papers on the topic. Usually, it's done in the context of manually adding pilots through special payload. You'd have to search on Google Scholar for example.

namyh18, 2019-11-29:
Thanks for your reply. 
I have another questions about  "wifi_loopback.grc".

1. Does bandwidth in "WiFi PHY Hier" is related to sampling rate? I cannot find any sampling rate in this loopback file. Or should I just connect throttle block in front of the Channel Model to adjust sampling rate?

2. To see the effect of the sampling frequency offset, should I change 'epsilon' variable? I thought that it changes resampling rate in "Polyphase Arbitrary Resampler" block and give sampling offset. 

3. This question is an extension of previous Q.3. I'm trying to give custom data bits directly in the "wifi_phy_hier.grc". To do this, I have to remove "Message Strobe" and "WiFi MAC" block in "wifi_loopback.grc". Also, I found that in "WiFi Mapper", it scrambles and interleaves data bits. I don't want this to happen. Therefore, I just want to give custom data+signal field before "OFDM Carrier Allocator". 
Do you think this implementation will work? Otherwise, can you give some insights to send identical data symbols in the loopback/wifi_phy_hier?

4. In "wifi_phy_hier.grc", OFDM Carrier Allocator has 'sync words' and last part of this 'sync words' is long training sequence. Then what is the rest of the part? Is this short training sequence?

Sorry for bothering you. I'm new to this field and gnuradio.

bastibl, 2019-12-01:
1: The PHY needs to know the bandwidth to map frequency/sample rate offsets to Hz. For the simulations (i.e., the loopback flowgraph) this is not very relevant. Do not use a Throttle block, just use the flowgraph as is.

2: Yes, change the epsilon. Since WLAN is supposed to derive the frequency and the sample clock from the same oscillator, it change both similarly. That means it doesn't just change the sample clock offset, but also the frequency offset.

3: Yes.

4: Yes, it's the short and the long training sequence. It's a slightly weird format, since the symbols are going through the cyclic prefix block afterwards... There are several issues about the topic, in case you wonder about the strange format.

5: Yes, you have to change the sync words. See the `utils/sync_words.py `script to see how the current ones are generated.

namyh18, 2019-12-04:
Thanks again.
I think this might be the last question.

So is there no sampling rate in loopback flowgraph? I think there should be some rate to sample carrier symbols.

bastibl, 2019-12-04:
I'm not sure what you mean with sample rate.

There is a common misconception about sample rate in SDRs. For the flowgraph, everything is just a stream of samples that it processes as fast as it can. In a simulation (like in the loopback flowgraph), it doesn't matter if it takes a millisecond or a minute to process the samples of a frame. That means there is no "sample rate" in terms of samples that are processed per second.

Yet, the samples represent a physical signal. The way the physical signal is mapped to samples, corresponds to a "sample rate". In the WLAN receiver, it assumes that the signal is sampled with the bandwidth of the signal (20MHz signal -> 20Msps sample rate, 10MHz signal -> 10Msps sample rate).

namyh18, 2019-12-08:
Thank you very much for your help.

closed 2019-12-08 by namyh18

nardvark, 2020-06-16:
> 
> 
> Thanks for your reply.
> I have another questions about "wifi_loopback.grc".
> 
>     1. Does bandwidth in "WiFi PHY Hier" is related to sampling rate? I cannot find any sampling rate in this loopback file. Or should I just connect throttle block in front of the Channel Model to adjust sampling rate?
> 
>     2. To see the effect of the sampling frequency offset, should I change 'epsilon' variable? I thought that it changes resampling rate in "Polyphase Arbitrary Resampler" block and give sampling offset.
> 
>     3. This question is an extension of previous Q.3. I slightly change the code of mapper.cc, not to interleave and scramble data_bits. Is this the right way to give identical data symbols?
>        I checked that when I'm doing this, I get same equalized carrier symbols at every ofdm symbols.
>        (for example, I give 90 "x"s through BPSK 1/2, 12~41 ofdm symbols has same order of equalized carrier symbols after channel estimation in frame_equalizer)
> 
>     4. In "wifi_phy_hier.grc", OFDM Carrier Allocator has 'sync words' and last part of this 'sync words' is long training sequence. Then what is the rest of the part? Is this short training sequence?
> 
>     5. How can I change LT field? I think I should change (1)  OFDM Carrier Allocator -> sync word part, (2) base::LONG, (3) sync_long::LONG, but (3) is complicated
> 
> 
> Sorry for bothering you. I'm new to this field and gnuradio.

If one wanted to create an ideal replica of the original I/Q that came in, they would need to get the ?state? information from descramble in decode_mac, then propagate it forward to mapper and use that info in the scrambler?

bastibl, 2020-06-16:
Not sure I get it... The discussion you cited was about the loopback flowgraph. In that case, you already have the original IQ signal. In a real-scenario, you need the scrambler value, the MCS, and the original data. The whole point of the receiver is to get that out of the noisy signal. To do this, it already uses the scrambler.

nardvark, 2020-06-16:
Right, I have the original IQ, but I want to build non-noisy IQ replica, to
use as a matched filter to use on the original IQ, because I want to make
some measurements on the raw IQ in the presence of some interference.  But
if I try to build a packet, I think the scrambling/interleaving turns my
matched filter into something less useful.

bastibl, 2020-06-17:
Did you delete your initial question...? This, again, doesn't make any sense to me. You have the the raw IQ samples without noise. And now you're looking for a complicate way to calculate the very same thing? Can't you add at least some sentences and explain what you're actually trying to do... Anyway, yes, if you fix the scrambler seed and the MCS, the TX part will generate the same frame. There's no option for this though, so you'll have to change the code.

nardvark, 2020-06-17:
Sorry for the confusion on the deleted comment.  After I posted it, I
decided that someone else's closed comment on the loopback example wasn't
the best place to post the question, and was going to reach out directly
later, but you had already received the notification and responded.

We have a receive-only application, where I want to receive wifi packets
and perform power estimates on them in the presence of interference.  So
what we would like to do is the following:
1. Receive via USRP
2. Detect packets in sync short, keep copies of raw I/Q for later
comparison.  We have done this by changing sync short to attach the raw IQ
to a tag.
3. Proceed as normal through sync long, frame equalization, etc...We made
minor modifications to these blocks to propagate our tag with the raw I/Q.
4. The decoded bytes are sent to a hierarchical block (similar to your
wifi_phy_hier, but with mapper slightly modified to accept / propagate our
raw I/Q tag) to make the matched filter.  The matched filter should be an
ideal (noiseless) version of the raw I/Q we saved off in sync short.  The
easiest way to accomplish this seemed to be to take the result of
decode_mac and pipe it into this hierarchical block to reconstruct the raw
I/Q.  But the mapper block normally applies scrambling, puncturing,
interleaving, etc...It wasn't clear to me if just disabling those steps
would result in an ideal replica of the detected I/Q, or if we need the
scrambler to operate with some information from the descramble operation in
decode_mac (probably propagated as another tag) to put the bits back in the
same order as they were initially received.
5. Use the ideal I/Q as a matched filter on the raw I/Q.  Now we can
perform an estimate of how much power is from the source associated with
the decoded packet, instead of a mix of it and other traffic/interference
that is overlapping in time.

Unfortunately I can't be more specific about the end application without a
release/publication approval (which I am hoping to get eventually).  I will
be sure to cite your paper/repo if we do publish, you've done a lot of
great work.

Best,
Dan

bastibl, 2020-06-17:
Thanks, now I get it :-) Not sure how exactly you add all samples as tag in Sync Short, but that's something that seems to be solved.  Regarding interleaving etc: the scrambler is the only non-deterministic part. So you only have to know the data payload, the MCS, and the scrambler seed. I'd add the scrambler seed as metadata in the receiver, just like it already annotates (encoding, freq_offset, snr). After the frame is decoded, you could send the PDU to a transmitter (directly to the PHY without the MAC). You'd have to extend the WiFi Mapper to check for an encoding tag (and set the MCS accordingly) and a scrambler tag (and fix the seed accordingly). The resulting frame should be a perfect copy of the transmitted waveform.