bastibl/gr-ieee802-11 #200: wifi_transceiver.py import error

khan2018, 2019-11-25:
Hi,

I have tested the wifi_tx / rx and loopback codes. 
But I have this error in wifi_transceiver,py.

When I run the nic.sh script I get the following error:
Traceback (most recent call last):
  File "./wifi_transceiver.py", line 34, in <module>
    from wifi_phy_hier import wifi_phy_hier  # grc-generated hier_block
ImportError: No module named wifi_phy_hier

When I run the grc file I get the following error:

Traceback (most recent call last):
  File "/home/ubuntu/Desktop/wifi_transceiver.py", line 357, in <module>
    main()
  File "/home/ubuntu/Desktop/wifi_transceiver.py", line 345, in main
    tb = top_block_cls()
  File "/home/ubuntu/Desktop/wifi_transceiver.py", line 245, in __init__
    self.blocks_tuntap_pdu_0 = blocks.tuntap_pdu('tap0', 440, False)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", line 2229, in make
    return _blocks_swig5.tuntap_pdu_make(dev, MTU, istunflag)
RuntimeError: gr::tuntap_pdu::make: tun_alloc failed (are you running as root?)

I would be grateful for your help!

bastibl, 2019-11-26:
You have to build the PHY block first. Open the corresponding flow graph in GRC and click build. See the readme.

khan2018, 2019-11-26:
I did build the phy layer first then I tried to run the nic.sh script.

bastibl, 2019-11-26:
But the PHY hier block cannot be found... Try running the loopback flowgraph from the console and see if it can find the PHY. Maybe it's an issue of the GRC environment vs your console environment. Usually the phy hier block is at `~/.grc_gnuradio` or where ever `${GRC_HIER_PATH}` points to. Make sure it's there. The second error is normal as the tun/tap device does not exist, i.e., was not created by the nic.sh script.

khan2018, 2019-11-26:
When I run as root :
python wifi_loopback.py 

I get the error :
Traceback (most recent call last):
  File "wifi_loopback.py", line 35, in <module>
    from wifi_phy_hier import wifi_phy_hier  # grc-generated hier_block
ImportError: No module named wifi_phy_hier

However, without root it works fine.

When I run as root,  python wifi_transceiver.py

I get the error
 
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.009.006-0-g122d5f8e

Traceback (most recent call last):
  File "wifi_transceiver.py", line 34, in <module>
    from wifi_phy_hier import wifi_phy_hier  # grc-generated hier_block
ImportError: No module named wifi_phy_hier

When I run without root,   python wifi_transceiver.py

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.009.006-0-g122d5f8e

MAPPER: encoding: 0
set_min_output_buffer on block 9 to 397056
set_min_output_buffer on block 11 to 397056
set_min_output_buffer on block 13 to 397056
set_min_output_buffer on block 14 to 397056
set_min_output_buffer on block 16 to 397056
-- Detected Device: B210
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Asking for clock rate 40.000000 MHz... 
-- Actually got clock rate 40.000000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
-- Detected Device: B210
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Asking for clock rate 40.000000 MHz... 
-- Actually got clock rate 40.000000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
set_min_output_buffer on block 32 to 100000
Traceback (most recent call last):
  File "wifi_transceiver.py", line 357, in <module>
    main()
  File "wifi_transceiver.py", line 345, in main
    tb = top_block_cls()
  File "wifi_transceiver.py", line 245, in __init__
    self.blocks_tuntap_pdu_0 = blocks.tuntap_pdu('tap0', 440, False)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", line 2229, in make
    return _blocks_swig5.tuntap_pdu_make(dev, MTU, istunflag)
RuntimeError: gr::tuntap_pdu::make: tun_alloc failed (are you running as root?)

bastibl, 2019-11-26:
You should not run anything as root. Just run the nic.sh script without root.. It will use `sudo` for the commands that require root access. And it will setup the device and start the flowgraph without root permissions...

bastibl, 2019-11-26:
...and if you don't actually need the tun/tap interface, just disable the block and start the flowgraph as normal user, without using the nic.sh script.

khan2018, 2019-11-26:
I did that now the console is just stuck on :
set_min_output_buffer on block 32 to 100000

khan2018, 2019-11-26:
Do we expect a similar output to the wifi_loopback or wifi_rx simulation?

bastibl, 2019-11-26:
I don't know what you are doing or what you are trying to achieve. So it's hard to tell what you should expect. Also wifi_rx is no simulation. It receives frames over the air.

khan2018, 2019-11-27:
Yes simulation wasn't the correct word. I meant I can receive frames using wifi_tx and wifi_rx but i cannot run the transceiver example with and without the nic script.

I am interested in finding out the EVM and constellation of the transmitted signals. I want to transmit at a slightly different frequency than the one in wifi standards so I manually edited the freq and sampling rate variables and added my desired frequency but for some reason I cannot receive on that frequency.

Also is it possible to find the constellation of a non-wifi standard and the EVM ?
I checked we have gr-mer which gives the modulation error rate but they use WX Gui than QT .
I am fairly new to gnu so I am still trying to figure lots of things.

Thanks,

bastibl, 2019-11-27:
If you use an SDR as sender and receive, you can change the sample rate and the frequency on both ends and it should work. If it doesn't you might have introduced a bug or used very extreme values.

If you want to use the transceiver (and do not use TUN/TAP) you can delete or disable the block and start the flowgraph as a normal user. If you have a problem, what's the exact error?

I don't get the question about the measurements: You have access to all subcarriers in the frame equalizer and you have the constellations after equalization. They are also plottet in the GUI. Isn't this exactly the data you need for any further processing/evaluations you have in mind?

khan2018, 2019-11-27:
Thank you for your replies.

When I disable tun/tap block the simulation runs without errors but there is no output the console only shows 
set_min_output_buffer on block 32 to 100000

over and over again.

I have a signal which is not  a wifi standard but uses the wifi frequency 2.477 GHz and I want to find its constellation. I am looking for a solution for this.

bastibl, 2019-11-27:
It should print this several times and then stop. After that, the transceiver prints received frames (if you didn't change the flowgraph). Since you don't send WLAN frames, there is nothing to see.

Also just the fact that your technology also uses the 2.4GHz band doesn't mean that this transceiver is helpful in any way. If your signal is not very close to WLAN, it's probably not.

closed 2019-12-21 by bastibl