bastibl/gr-ieee802-11 #201: cant sniff self on yours scheme

FatboyShit, 2019-12-08:
i get it https://www.bastibl.net/gnuradio-wlan-scapy/
and done it 
[image]
but cant see frame ,and cant see in devices!!! what the shit???

bastibl, 2019-12-08:
First of all, remove the RX part at the bottom. Stick to the simple flow graph shown in the blog post. This might just add problems. Once the flowgraph starts up, make sure to switch to 20MHz sample rate and tune to a WLAN channel (11a or 11g, _not_ 11p). Then start airmon or whatever you use and listen on the configured channel. You can also just switch the WLAN card in monitor mode and use Wireshark.

ValZapod, 2019-12-08:
You cannot say "what a shit", it is wrong in English, but you can say what THE fuck or even what the fucking shit you are talking about, etc. Also, I will recommend this to you, a video with very strong bad language which is also word for word from a real transcript, you can set with the pdf https://www.docdroid.net/VuC6p0Y/transcript-denver-allen.pdf and check it out))  https://youtu.be/7vN_PEmeKb0

FatboyShit, 2019-12-08:
nothing no work 
[image]

marcusmueller, 2019-12-08:
@FatboyShit that post comes with an exact flow graph to work with, and as far as I can tell it's not really @bastibl to do your work / write your thesis / do your contracting, unless he's agreed to you're paying him an hourly wage that reflects the amount of experience and work that went into writing gr-ieee802_11.

FatboyShit, 2019-12-08:
> @FatboyShit that post comes with an exact flow graph to work with, and as far as I can tell it's not really @bastibl's job to do your work / write your thesis / do your contracting, unless he's agreed to you paying him an hourly wage that reflects the amount of experience and work that went into writing gr-ieee802_11.

dont worry animalfucker

bastibl, 2019-12-09:
@FatboyShit This is a physical layer implementation of WLAN (layer 1). The actual social and technical issues here seems to be on layer 8. I'll close this.

closed 2019-12-09 by bastibl