bastibl/gr-ieee802-11 #204: wifi_rx cannot decode frame

HuangZengPei, 2019-12-20:
Hi Bastian,
First of all, thank you so much for your contribution
I want to build the communication between two USRPs, but I met the same problem as in issue #186. My environment is as follow:
Ubuntu 18.04, GNU radio 3.7,USRP N210 and SBX daughterboard.
I can correctly run the wifi_lookback.grc in one USRP and see the decoded frame, but in the Rx, terminal shows "long :frame start at 320". Tx and rx are in the same channel.
The lookback looks like this
[image]
and these are tx and rx
[image]
[image]
I am new to USRP and I wonder how to solve it.

bastibl, 2019-12-20:
What are exactly are you trying to do here? Do you have two  SDRs connected to the same PC (there seems to be no configuration to address a particular device)? If so try on different PCs. Or better start with a simpler setup (i.e., receiving frames from a WLAN card). Also check gain, antenna connections, sample rate, etc

HuangZengPei, 2019-12-20:
Thanks for your reply. I want to use one USRP as wifi_tx and another as wifi_rx. The two USRPs are connected to different PCs. But neither PC has a WLAN card. Does it a problem? Or is it caused by the interference of other WiFi signals? Moreover, if the terminal output "D", does it mean USRP received OFDM frames?

bastibl, 2019-12-22:
It is very hard to help, since you seem to have not much experience and don't provide a lot of information. In general, there are hundreds of things you could have done wrong. Also you didn't install your UHD drivers correctly (see the warnings). Also the "D" come from the driver of you SDR, indicating that the PC cannot keep up with the samples. You should really read the manual of your SDR driver. It's not possible to list everything you could have done wrong. And you still didn't say what you have already checked. Also you start with a rather complex setup, making it even harder to debug. Why don't you just try to decode frame from a normal WLAN card to understand if the RX or the TX causes the problem. You obviously don't need a  WLAN card to run GNU Radio. So here are some ideas: 
- install the driver correctly
- make sure there is no switch or hub between the SDR you NIC
- don't run in a VM
- try different gain setting
- check that the antenna is connected to the correct port
- use a PC that manages the sample rate (i.e., doesn't print "D"s)

HuangZengPei, 2020-01-07:
Hi Bastian,
    Thank you for all the help you have given me with my work. I have received and decoded OFDM frames correctly.  But I have some confusion about CSI.
   I successfully modified ls.cc and got CSI, when I run wifi_lookback.grc in one PC, the CSI looks like(46.0934,4.01396), but when I run wifi_tx and wifi_rx on different PC with two USRPs, the CSI value is (0.00160867,0.00384978). The two USRPs are close to each other. I know you once said both I and Q components have to be in the range [-1, 1], but is it too small?
   Thanks a lot for your help.

bastibl, 2020-01-08:
The SDR always outputs floats in the range [-1;1]. If you send a signal with higher amplitude it will clip. An OFDM signal has a high PAPR, so you want the average amplitude to be way below 1.

closed 2020-04-13 by bastibl

ChengDaHaI, 2020-08-09:
> Hi Bastian,
> Thank you for all the help you have given me with my work. I have received and decoded OFDM frames correctly. But I have some confusion about CSI.
> I successfully modified ls.cc and got [...]

Hi,

I face the same  problem where rx terminal shows "long :frame start at 320". Can you explain how do you solve it? 

Thanks.

bastibl, 2020-08-12:
...and what exactly is your setup. Also, did you consider the recommendations in this issue, the readme, and other related issues?

ChengDaHaI, 2020-08-12:
> ...and what exactly is your setup. Also, did you consider the recommendations in this issue, the readme, and other related issues?

Hi Bastible,

Finnaly i solve the issues by
1. set the two parameters in FER correction of USRP source to True
2. specifiy the antenna to TX/RX of both USRP sink and source (the default is not specified).