bastibl/gr-ieee802-11 #208: module 'ieee802_11' has no attribute 'BPSK_1_2' (also 'LS' and 'signal_field')

gefa, 2020-01-16:
Hi Bastian,

As the title says parameters ieee802_11.BPSK_1_2, ieee802_11.LS and alike from cannot be evaluated. Therefore wifi_phy_hier.grc cannot be built. I am not saying gr-ieee802-11 is responsible for this error, but in case you have some input I'm all ears.

Full error:
```
grcc examples/wifi_phy_hier.grc
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
/home/user/.grc_gnuradio
/usr/share/gnuradio/grc/blocks
/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/user/workspace/gr-ieee802-11/examples/wifi_phy_hier.grc
ERROR: gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block chan_est
Traceback (most recent call last):
File "/usr/lib/python3.8/site-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
value = eval(parameter_block.params['value'].to_code(), namespace)
File "<string>", line 1, in <module>
AttributeError: module 'ieee802_11' has no attribute 'LS'
```
My environment:
archlinux 5.4.11-arch1-1 ... x86_64 GNU/Linux
GNU Radio Companion 3.8.0.0 (Python 3.8.1) -- installed with pacman package manager
gr-ieee802-11 and gr-foo installed as instructed in readme

cmake seems to finish fine (I doubt that gmp and mpir have anything to do with this, I have installed them, but maybe cmake is not finding them...)
```
cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr/lib ..
-- Build type not specified: defaulting to release
-- Checking for module 'gmp'
--    Package 'gmp', requested by 'virtual:world', not found
-- Checking for module 'mpir >= 3.0'
--    Package 'mpir', requested by 'virtual:world', not found
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/libpython3.8.so (found suitable exact version "3.8.1")
-- Extracting version information from git describe...
-- Using install prefix: /usr/lib
-- Building for version: 99f1a456 / 1.1.0.git
--
-- Checking for module SWIG
-- Found SWIG version 4.0.1
-- Found PythonLibs: /usr/lib/libpython3.8.so (found version "3.8.1")
-- Configuration done
-- Generating done
-- Build files have been written to: /home/user/workspace/gr-ieee802-11/build
```
But gr-ieee802-11/build/CMakeFiles/CMakeError.log looks worrisome:
```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/gefa/workspace/gr-ieee802-11/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_67d87/fast && /usr/bin/make -f CMakeFiles/cmTC_67d87.d
ir/build.make CMakeFiles/cmTC_67d87.dir/build
make[1]: Entering directory '/home/gefa/workspace/gr-ieee802-11/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_67d87.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD   -std=gnu11 -o CMakeFiles/cmTC_67d87.dir/src.c.o   -c
/home/gefa/workspace/gr-ieee802-11/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_67d87
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_67d87.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    CMakeFiles/cmTC_67d87.dir/src.c.o  -o cmTC_67d87  
/usr/bin/ld: CMakeFiles/cmTC_67d87.dir/src.c.o: in function `main':
src.c:(.text+0x3e): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x4a): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x5b): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_67d87.dir/build.make:87: cmTC_67d87] Error 1
make[1]: Leaving directory '/home/gefa/workspace/gr-ieee802-11/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_67d87/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
 return data;
}

int main(void)
{
 pthread_t thread;
 pthread_create(&thread, NULL, test_func, NULL);
 pthread_detach(thread);
 pthread_join(thread, NULL);
 pthread_atfork(NULL, NULL, NULL);
 pthread_exit(NULL);

 return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following ou
tput:
Change Dir: /home/gefa/workspace/gr-ieee802-11/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_a8364/fast && /usr/bin/make -f CMakeFiles/cmTC_a8364.d
ir/build.make CMakeFiles/cmTC_a8364.dir/build
make[1]: Entering directory '/home/gefa/workspace/gr-ieee802-11/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_a8364.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -std=gnu11 -o CMakeFiles/cmTC_a8364.dir
/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_a8364
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_a8364.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_a8364.dir/CheckFunction
Exists.c.o  -o cmTC_a8364  -lpthreads  
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_a8364.dir/build.make:87: cmTC_a8364] Error 1
make[1]: Leaving directory '/home/gefa/workspace/gr-ieee802-11/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_a8364/fast] Error 2
```
Looking at this CMakeError, I'm under the impression that something could be missing from CMakeLists.txt or could it be a cmake bug?

I have seen several related issues but they don't quite apply to my case I believe:
https://github.com/bastibl/gr-ieee802-11/issues/192
https://github.com/bastibl/gr-ieee802-11/issues/6
https://www.ruby-forum.com/t/asking-help-about-gr-ieee802-11/240300/3

Thanks in advance,
Stefan Gvozdenovic

gefa, 2020-01-17:
I have worked around this issue for now by installing gr-ieee-80211 with pybombs. Not sure what was off with the original issue, but I'll close this.

closed 2020-01-17 by gefa

jkwcht, 2020-01-17:
I had the exactly same problem even when installing gnuradio and gr-ieee-80211 by pybombs. The Issue was that the python package 'ieee802_11' was not included in 'setup_env.sh' of the related pybombs prefix. The package is located at /pathToPybombsPrefix/lib/python3/dist-packages

jscatena88, 2020-03-06:
Is it possible to reopen this issue? I'm having the same issue on Manjaro. And would like to get everything working without pybombs

bastibl, 2020-03-09:
What exactly is your issue and what exactly did you do? This "issue" has been discussed many, many times. If you have a problem that has not been addressed, please add a new issue.
In that past, it was always one of:
- incompatible versions of GNU Radio and this module
- the install path of the Python module was not in the PYTHONPATH
- the install path of the shared library was not in LD_LIBRARY_PATH
- the shared library was installed in a system directory and `sudo ldconfig` was not run