bastibl/gr-ieee802-11 #210: Buffer too small for min_noutput_items

sbindel, 2020-01-27:
Hi Bastian,
During the test of the loopback example, I have observed when I increase the PDU length an error related to the IEEE802.11 chuncks to Symbols block indicating :  
**thread[thread-per-block[10]: <block chunks_to_symbols(7)>]: Buffer too small for min_noutput_items** ,when I use BPSK 1/2 modulation.

However when I used a QAM module the following error appears : 
**<block ofdm_cyclic_prefixer(11)>]: Buffer too small for min_noutput_items**

Is it a problem related to GR_FIXED_BUFFER_SIZE ? Have I to modify buffer of each block ?

Best regards

sbindel, 2020-01-28:
I have set the GR_FIXED_BUFFER_SIZE to 128k, by compiling from the source GNU Radio and it's work.
Is there a way to change the  GR_FIXED_BUFFER_SIZE variable without compile manually ?

bastibl, 2020-01-29:
Please see this thread and make sure to use the most recent GNU Radio version. https://github.com/bastibl/gr-ieee802-11/issues/190

closed 2020-04-13 by bastibl