bastibl/gr-ieee802-11 #211: wifi_phy_hier errors

msfu101, 2020-02-15:
hi there,
under Kali Linux and Gnuradio 3.8.0 i built gr-foo and gr-ieee802-11 from Source with no Errors. Then i opened the wifi_phy_hier.grc to build the necessary blocks but i can't build it.
The Error Messages are in the attached file.

[image]

thanks in advance

bastibl, 2020-02-17:
Please make sure that the path of the Python modules is in our PYTHONPATH variable. In your case, it seems to be `/usr/local/lib/python3/dist-packages`.

msfu101, 2020-02-22:
I set the PYTHONPATH Variable and now... it works!
Thank you

closed 2020-02-22 by msfu101

Cormallonal, 2020-04-03:
Hello, I have the same problem. However, I don't know where I can find the PYTHONPATH variable. Could someone briefly explain to me how and where I can find and change them?

bastibl, 2020-04-03:
Sorry, but how you set an environment variable is not specific to this module and it wouldn't make a lot of sense to explain it the 43.800.001st time. Please google it.