bastibl/gr-ieee802-11 #212: Examples not running

TheL1ne, 2020-03-09:
Hello,
again someone really new to the USRP world here. I did just set up a USRP N210 and installed everything (drivers, gnu-radio, GRC and your work) from scratch and was now trying to run your examples. The loopback works just fine but the `wifi_rx.grc` and the `wifi_transcieve.grc` load just fine but when running them result in this error: 
```
[INFO] [USRP2] Current send frame size: 1472 bytes
Traceback (most recent call last):
  File "/home/noah/gitprojects/gr-ieee802-11/examples/wifi_rx.py", line 373, in <module>
    main()
  File "/home/noah/gitprojects/gr-ieee802-11/examples/wifi_rx.py", line 361, in main
    tb = top_block_cls()
  File "/home/noah/gitprojects/gr-ieee802-11/examples/wifi_rx.py", line 150, in __init__
    self.uhd_usrp_source_0.set_auto_dc_offset("", 0)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 3499, in set_auto_dc_offset
    return _uhd_swig.usrp_source_sptr_set_auto_dc_offset(self, enb, chan)
TypeError: in method 'usrp_source_sptr_set_auto_dc_offset', argument 2 of type 'bool'
```
These are the lines for `wifi_rx.grc` but pretty much the same happens on the other example too. Is there a way to fix this?

bastibl, 2020-03-09:
Please see this issue: https://github.com/bastibl/gr-ieee802-11/issues/157

TheL1ne, 2020-03-10:
Ah Thanks. I thought that I was probably not the first one to ask, but I was unable to find a similar issue.

TheL1ne, 2020-03-13:
I am sorry but I think I need to reopen this ticket because I am stuck again at the tranceiving example `wifi_transceiver.grc`:
```
Traceback (most recent call last):
  File "/home/noah/gitprojects/gr-ieee802-11/examples/wifi_transceiver.py", line 361, in <module>
    main()
  File "/home/noah/gitprojects/gr-ieee802-11/examples/wifi_transceiver.py", line 349, in main
    tb = top_block_cls()
  File "/home/noah/gitprojects/gr-ieee802-11/examples/wifi_transceiver.py", line 246, in __init__
    self.blocks_tuntap_pdu_0 = blocks.tuntap_pdu('tap0', 440, False)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig6.py", line 1046, in make
    return _blocks_swig6.tuntap_pdu_make(dev, MTU, istunflag)
RuntimeError: gr::tuntap_pdu::make: tun_alloc failed (are you running as root?)
```
I saw that this error was mentioned in https://github.com/bastibl/gr-ieee802-11/issues/173 but I was unable to find a solution in there and running it as root caused priviledge problems:
```
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x140
```

So as mentioned in the other issue: running it as root does not seem to be a valid apporach.

Btw.: the example `wifi_rx.grc` works now.

bastibl, 2020-03-13:
Please see these issues and coments:
https://github.com/bastibl/gr-ieee802-11/issues/102#issuecomment-368572726
https://github.com/bastibl/gr-ieee802-11/issues/200#issuecomment-558635537
https://github.com/bastibl/gr-ieee802-11/issues/194#issuecomment-544859815
https://github.com/bastibl/gr-ieee802-11/issues/53

TheL1ne, 2020-03-16:
Okay I did setup everything according to `nic.sh` script (copied the setup to a terminal) which worked fine but now the transceiver example stopped at:
```
set_min_output_buffer on block 32 to 100000
gr::tuntap_pdu: failed to set MTU to 440.
You should use ifconfig to set the MTU. E.g.,
  $ sudo ifconfig tap0 mtu 440

Allocated virtual ethernet interface: tap0
You must now use ifconfig to set its IP address. E.g.,
  $ sudo ifconfig tap0 192.168.200.1
Be sure to use a different address in the same subnet for each machine.

set_min_output_buffer on block 35 to 100000
```
And running  `sudo ifconfig tap0 mtu 440` resulted in the same error as above again. I feel like I am missing something vital here. Even running the `nic.sh` itself resulted in this error now.

bastibl, 2020-03-16:
What do you mean with it "stopped." Did it crash? I'd think that these messages are just warnings. Setting something on the interface (within GNU Radio) is supposed to fail, since you configure it manually. Also, what's the MTU of the interface? Isn't it already 440?

TheL1ne, 2020-03-21:
It did stop and the window to display the example would not appear. But now some days later I did ran the `nic.sh` script again and now the tranceiver example works too :+1:

closed 2020-03-21 by TheL1ne