bastibl/gr-ieee802-11 #213: Wrong Checksum in "WiFi Decoder MAC"

TheL1ne, 2020-04-06:
I try to run a 2 machine setup with 2 Ettus Research N210 modules (one on each PC). 
Configuration is:
```
|       Device: USRP2 / N-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: N210
|   |   hardware: 2561
|   |   mac-addr: a0:36:fa:25:32:7d
|   |   ip-addr: 192.168.10.2
|   |   subnet: 255.255.255.255
|   |   gateway: 255.255.255.255
|   |   gpsdo: none
|   |   serial: E7R12XDUP
|   |   FW Version: 12.4
|   |   FPGA Version: 11.1
|   |   
|   |   Time sources:  none, external, _external_, mimo
|   |   Clock sources: internal, external, mimo
|   |   Sensors: mimo_locked, ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -50.000 to 50.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   
|   |   |   Freq range: -50.000 to 50.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |   ID: XCVR2450, XCVR2450 - r2.1 (0x0061)
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: 0
|   |   |   |   Name: XCVR2450 RX
|   |   |   |   Antennas: J1, J2
|   |   |   |   Sensors: lo_locked, rssi
|   |   |   |   Freq range: 2400.000 to 6000.000 MHz
|   |   |   |   Gain range LNA: 0.0 to 30.5 step 15.0 dB
|   |   |   |   Gain range VGA: 0.0 to 62.0 step 2.0 dB
|   |   |   |   Bandwidth range: 13500000.0 to 39600000.0 step 600000.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: ads62p44
|   |   |   |   Gain range digital: 0.0 to 6.0 step 0.5 dB
|   |   |   |   Gain range fine: 0.0 to 0.5 step 0.1 dB
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -200.000 to 200.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |   ID: XCVR2450 (0x0060)
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: 0
|   |   |   |   Name: XCVR2450 TX
|   |   |   |   Antennas: J1, J2
|   |   |   |   Sensors: lo_locked
|   |   |   |   Freq range: 2400.000 to 6000.000 MHz
|   |   |   |   Gain range VGA: 0.0 to 30.0 step 0.5 dB
|   |   |   |   Gain range BB: 0.0 to 5.0 step 1.5 dB
|   |   |   |   Bandwidth range: 24000000.0 to 48000000.0 step 12000000.0 Hz
|   |   |   |   Connection Type: QI
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: ad9777
|   |   |   |   Gain Elements: None
```
I try to run the wifi_tx and wifi_rx examples and now debugged my way into the `WiFi Decoder MAC` and it prints out that it got a wrong checksum which may be caused by a misinterpretation of the encoding parameter as prints as debug: `Decode MAC: frame start -- len 786  symbols 33  encoding 6`.
when I did run everything with default values so encoding should be `0` which refers to `BPSK 1/2` if I am not mistaken.

Is this an issue on my end or is this really a bug? I did build everything from source using the `maint-3.7` branches on both machines.

TheL1ne, 2020-04-06:
I just reset my branches because I tried working around this issue and with no changes at all (except the ones from #212 ) the frames seem to contain fluctating encoding parameters. I got sometimes `2`, `7` or `4`.

bastibl, 2020-04-07:
It's hard to track it down with just limited information, but some ideas/hints:

- even if something goes wrong at the early stages of the receiver, it will (most likely) be dropped at the CRC. That does *not* mean that it's a problem of the decoder or the CRC implementation. Since you see (seemingly) bogus data in the signal field, it looks like there is a problem at the earlier stages.
- Since you sent the frames yourself, you should know what exactly you'd expect in the signal field. If whatever you decode doesn't match your configuration, you have a problem and later stages have no chance to decode anything, i.e., the frame will inevitably be dropped after the CRC check.
- To better understand the problem, I'd just tests RX and TX with normal WLAN cards in order to understand if your problem is RX or TX.

TheL1ne, 2020-04-07:
I found the error: I did not configure the antenna to use as I only have an antenna attached at one of the antenna ports for each machine. It works if I set this correctly in the USRP source and sink.

closed 2020-04-07 by TheL1ne