bastibl/gr-ieee802-11 #216: Use wifi_rx.grc to capture IQ data comes from wifi transmitter

p577665228, 2020-04-16:
Hi Bastian,

I use wifi_rx.grc to capture IQ data comes from wifi transmitter. So I add a file_sink module after UHD_USRP_Source module. But the captured IQ data comes from transmitter, receiver, and blank(note: if that time has no communication, the IQ data captured will be blank).

 How can I save the IQ data just comes from transmitter?

bastibl, 2020-04-16:
Does your USRP work in general? Can you decode the signals with the unmodified flowgraph (if not, you have another problem)? Are there warning about drops or driver issues? How do you open the file and inspect it? What do you mean with "blank", zeros or small values? Can you decode the samples from the file, if  you pipe them into the receiver? etc etc

I'm not sure if I understand your question. Directly after the USRP you get all samples. Another option is to record the output of Sync Short. This is the data after frame detection, which might miss the short preamble. Everything else is not supported out of the box, but you could just use a simple energy detector, for example.

Anyway, I'm not sure if this is really related to this module, since you don't use any of its functionality. As far as I understand, you only use USRP->File Sink.

p577665228, 2020-04-17:
yes,  my usrp can work. And It can decode wifi signal. But some IQ sample can't be decoded (frame is too short to decoded). I want to save IQ sample which can be decoded.

p577665228, 2020-04-17:
My usrp can work. And it can decode wifi signal. Your suggest is right, after Sync Short, the IQ data can be decoded.
But the caputure IQ data  after Sync Short come from transmitter and receiver. How can I just save the IQ data just come from transmitter?

bastibl, 2020-04-17:
I'm not sure, I understand you correctly. But what you want does not work out-of-the-box and is also not trivial to implement. You have to fully decode the signal to know who send it. This is the last stage of the receiver. The IQ baseband samples are at the very start. 
The easiest way might be to just record all samples and extract frames from a particular receiver offline.

p577665228, 2020-04-17:
Yes, I have saved the IQ data after Sync Short. 
Do you have any program to extract IQ data come from transmitter.

p577665228, 2020-04-17:
Yes, I have saved the IQ data after Sync Short.
Do you have any program to extract IQ data come from transmitter.
You can see that the IQ data from receiver is Intel..., and the info of reveiver's IQ data is just Qos Null function or Qos Data, so can you use this feature to filter the IQ data from the receiver?

[image]

bastibl, 2020-04-20:
I still don't know what exactly you want to do. Every time you ask a question, I'd have at least 10 follow-up question before I could give a meaningful answer. Again, my understanding is that you want to decode a frame and once you figured that it is from a particular sender, you want to store the IQ samples. As I already said, this is not possible out of the box and you would have to implement it.
If you want to do something completely different. Please put some effort in your questions and explain in more detail what you want to do.

p577665228, 2020-04-21:
Thank you ! You are right, I should do it by my self.

closed 2020-04-21 by p577665228