bastibl/gr-ieee802-11 #218: LimeSDR freezing GNU Radio

Formula007, 2020-05-10:
Hallo,
would you please help me with this project?
I have LimeSDR-USB and I am trying to catch (receive) communication between two mikrotik APs communicating on the 802.11g 20MHz standard. So 2,4GHz, because 5GHz is out of bandwith of LimeSDR.
[image]

The problem is that out of 100 packets sent, I capture approximately 3 of them.
[image]

The error seems to be in poor performance, as the log window is freezing. 
I have an i7 4 cores, 8GB RAM.
I have also tried to reinstall everything, also with another computer with same result.
I'm running Ubuntu 18.4.4, GNU Radio 3.7.11 in a virtual box, but I have the same problem in native Ubuntu.
Here I copied log in to txt file...
[log.txt](https://github.com/bastibl/gr-ieee802-11/files/4606015/log.txt)
Projet setting: 
lo_offset: 0
gain: 20 db
freq: 2.437GHz (chan 6)
samp_rate: 20MHz
I am using LimeSuite Source instead of USRP source.
[image]
I can also see, that when i add waterfall graph, tahat time is running faster in real life than in GNU Radio :D 
[image]
The log often says:
Warning: starting to receive new frame before old frame was complete.

I went through Troubleshooting and none of that helped me.
I have tried:
calibrate daughterboard
run volk_profile
different gain settings
real-time priority
tryed to tune the RF frequency out of the band of interest with LO offset set
connect you PC directly to the USRP without a switch in between
I have also LimeSDR mini and there is same problem and I also tried RFtap project, with same result.
[image]
Have anybodey some idea what else can I try?
Thank you guys :)

bastibl, 2020-05-11:
Some things you could check:
- The SNR doesn't look very good. Try to change the gain, check the antenna connector etc. It shouldn't look that cluttered.
- Use a normal WLAN card in monitor mode and check what kind of frames are exchanged. The receiver uses a hard-coded maximum frame size of 1500 byte. If it's bigger it will not be decoded.
- I don't have a LimeSDR, but you should check if it supports indications when samples get dropped. Since the CPU is close to max, it might well be that it looses samples.
- Use GNU Radio 3.8 on a native installation. Piping 20MHz of BW over USB into a VM can cause problems.
- Test a more capable PC

Formula007, 2020-05-11:
Thank you for fast reply.

- I tried all steps of gain :/

- I did the measurements in parallel with the wireless card and SDR at the same time...
[pcaps.zip](https://github.com/bastibl/gr-ieee802-11/files/4609044/pcaps.zip)
Frames are smaller than 1500 bytes.

- I am not sure, but I think there is no such indicators.

- OK I will try GNU 3.8, last time I had some problem with instalation of gr-limesdr module.

- Do you think that it could by caused by low power of PC? What configuration do you recomend?

Yesterday I've tried to save signal string directly from source to file sink and than from file to your flowgraph, but it also did not help.

bastibl, 2020-05-11:
Do the constellations always look so noisy? Is the transmitter far away? Is the antenna is connected at the correct port? Is the antenna compatible with the port (SMA vs RP-SMA).

I guess it's a combination of running in a VM and not enough CPU power. Just run it *natively* on a desktop or a decent laptop to make it's not the CPU.

I don't know how the Lime behaves when there are overruns (e.g., if there are overflow indicators through GNU Radio tags or output on the console.) The noisy constellations could also be caused by overruns. The USRPs, for example, restart streaming after an overruns, which causes transients. Something similar could happen here.

Formula007, 2020-05-14:
Yes, constellations are like that and sometimes even worse. 
I tryed to move transmiter closer and farther and I think that on opposite site of room is the best.
[image]

Today I tried to install Ubuntu on my notebook with GNU Radio 3.8.1 and results are same as before.. Everything was dropping.
[image]
According to htop there should be no performance limitations.

Even when I stopped the flowgraph, decoding in console ran for another 10 minutes.
[log z videa.txt](https://github.com/bastibl/gr-ieee802-11/files/4630874/log.z.videa.txt)

bastibl, 2020-05-18:
The autocorrelation looks like the device has an uncompensated (or not fully compensated) DC offset. For a quick test, you could try to increase the threshold of the Sync Short block. Also you might want to add a trigger at the autocorrelation plot (at the top) to find a good value for the threshold. You should clearly see the plateau of the autocorrelation at the beginning of a frame.

Formula007, 2020-05-26:
I tried to lower the threshold as you said, but nothing is changed and I am not sure which trigger do you mean.
I have also BladeRF 2.0, but it shows exactly the same. Each packet is dropped due to a bad checksum.
In addition, zeros appear in the console. What does this mean?
[image]

bastibl, 2020-05-27:
I asked you to **increase** the threshold. I also asked you to run it natively since USB emulation of VMs can cause all kinds of problems.

The zeros are probably "O"s and indicate that your PC cannot keep up with the data, which might be caused by running in the VM, an uncompensated DC offset, or something else. But please first try these two things first.

Formula007, 2020-05-29:
I am sorry, that was just type error, I increased threshold.
I ran linux natively and the result is still the same. I recorded the whole procedure on youtube.
https://www.youtube.com/watch?v=dtR28pm83o0
Then I increased threshold...
https://www.youtube.com/watch?v=zl6B3fWa838&t=100s
And still nothing.

bastibl, 2020-05-31:
Not sure what you're doing, but the terminal output looks like it several orders of magnitude too slow. It looks like it still outputs minutes after the flowgraph stopped...
Please
- check that the SDR is connected at a USB 3 port
- disable the debug output, you can also disable the Wireshark stuff and just keep the Parse MAC block
- use a different terminal emulator
- start the flowgraph directly in the terminal (not in GRC). You have to run the python file that GRC creates in the example directory.

closed 2020-07-30 by bastibl