bastibl/gr-ieee802-11 #219: Invisible in mobile

Venkyvulpine, 2020-05-11:
Hi,
The wifi created in router is not visible in mobile Do you know why?

bastibl, 2020-05-11:
How about adding a little bit of context?

Venkyvulpine, 2020-05-11:
Hi,
I am using your source code and build transmitter which transmits signal
and it is recieved by reciever (both are routers) but the wifi is not
showing up in mobile which is produced by transmitter router do you know
why it is not working? I have tried changing frequency and bandwidth
according to the mobile.mobile I have used is samsung galaxy s8.

Some of the doubt I have:
1. uhd doesn't have a name so is this impact on visibility.

Venkyvulpine, 2020-05-11:
Additional information:
We use router as USRP N210

Thanks
Venkatesh

closed 2020-05-11 by bastibl

Venkyvulpine, 2020-05-11:
Why you closed this?

bastibl, 2020-05-11:
Because there is no way to help you, if you don't manage to formulate a clear question.