bastibl/gr-ieee802-11 #220: Minimun processing power?

gmartina, 2020-05-12:
Hi Bastian,

Based on you experience, do you have any info regarding minimun processing power? 

Im am trying to run transceiver.grc in a Quad-core ARM Cortex-A53 1.2GHz 64 bit, but the processor cant keep with the program (>95%) and it crashes after 10 seconds. Before running it, the processor is idle 99% of the time, so there isnt anything bad.

So I conclude that cortex-a53 is not enough to run gr-ieee802.11. 

Is there anything else i can do to chek more things?

thank you

bastibl, 2020-05-12:
I'm not sure what you are measuring ad what you mean with ">95%". Is that the load of one core, the average load of all cores? Anyway, the program should never crash. If the CPU is not fast enough, it will drop samples and ultimately miss frames. But it should never crash.

Also, it does not make much sense to say "CPU X is not good enough for gr-ieee802-11". If it "works" depends on so many things, e.g.:
- sample rate / signal bandwidth
- is the channel idle or a frames sent back-to-back
- are there very short frames (i.e., much overhead from synchronization) or long frames with 64-QAM (i.e., much overhead from the decoder).
- GNU Radio versions
- VOLK profile results
and many many many more.

gmartina, 2020-05-22:
Hi Bastian.

One more question. My analog devices transceiver (ad9361) has a configurable hardware FIR filter. I would like to use it, if possible, to speed up the "examples/wifi_transceiver.grc". 

Could you  tell me where is the filtering part in your gr-ieee-802-11 implementation? So I could remove it and use the hardware FIR instead.

Do you think this is a good idea?

Thank you very much,

Gustavo

bastibl, 2020-05-25:
I'm not sure how you want to use the HW filter for this PHY. You cannot just replace a filter at some point in the flowgraph with the HW version (which would require streaming the samples back-and-forth).
AFAIK, the HW filter is relevant if you oversample the signal, e.g., if you sample with 20MHz but are only interested in 1MHz of the spectrum. In that case, you could use the HW filter to get rid of noise that is outside the band of interest.

closed 2020-05-27 by gmartina