bastibl/gr-ieee802-11 #221: usrp uhd device underruns

Venkyvulpine, 2020-05-13:
while testing  the device I found that it is actually underrunning 
what is the problem which cause device to underrun

For Your Information:
1. usrp device is directly connected to pc(switch is not used).
2. I am not  using virtual machine.
3.volk_profile  test is succesfull everything is in best state.

[image]

bastibl, 2020-05-13:
The PC cannot keep up with the data. Try a more powerful PC.

Venkyvulpine, 2020-05-13:
Thank you for your response!

Venkyvulpine, 2020-05-15:
Recommended Specification??

bastibl, 2020-05-18:
There is another issue about the specs. Also you opened another issue about a very related topic. Also there is no information about what you are doing or what hw, etc you are using. I close this.

closed 2020-05-18 by bastibl