bastibl/gr-ieee802-11 #223: Extract data as csv format from wifi_rx pcap capture

armada-h, 2020-05-15:
What i want to achieve : a csv output file like the one used for airodump-ng, to feed it to a gui interface for better visualization (Networks ,Client , Pies , Graphs ... ), in reel time.

What i did : 
* Launching wifi_rx from a shell script, works great
* Wireshark start automatically from the shell script 
* wifi.pcap is registered in /tmp/ directory

is there a way to extract data as .csv format from wifi.pcap in reel time (like bellow)

BSSID, First time seen, Last time seen, channel, Speed, Privacy, Cipher, Authentication, Power, # beacons, # IV, LAN IP, ID-length, ESSID, Key

i tried to add this to the shell script but no luck  `airodump-ng -r wifi.pcap -w wifi`
this instruction works fine when the script is stooped, it generate  csv file, but i want the csv file  to be generated it in reel time (appended) to see the evolution of the capture
System : Ubuntu Linux 18.04
Any suggestion @bastibl  . thank you

bastibl, 2020-05-18:
This can be done with `tshark` you can either replace Wireshark or use tshark to read the PCAP file. To get a CSV you have to do something like:

```
tshark -E separator=, -T fields -e wlan.sa -e wlan.da -r <file>`
```

Add fields as needed.

armada-h, 2020-05-18:
thanks @bastibl, it's works great

bastibl, 2020-06-22:
glad it works for you

closed 2020-06-22 by bastibl