bastibl/gr-ieee802-11 #226: Problem Wifi_Phy_Hier

oabravo, 2020-06-14:
I am having the next problem with the wifi_phy_hier and i can't run it
[image]
[image]
[image]
[image]
please i need help

bastibl, 2020-06-15:
This is very likely because either the python module is not found (it is not in the PYTHONPATH) or the python module cannot find the linked library because it is not in you LD_LIBRARY_PATH. You can open a python console and `import ieee802_11`. This might give more debug output.
Please check the installation instructions and the many issues on the topic.

oabravo, 2020-06-15:
Python console show me this
But i don't know how solve it, i tried adding the direction /usr/local/lib/python3/dist-package/ to the PYTHONPATH but it didn't work
[image]

bastibl, 2020-06-15:
Sorry, but you only provide minimal information. I don't know how or if you installed the module. Let alone where you installed it. You don't write if you actually checked that the files are actually in the folder that you cited. Anyway, since this was discussed in many, many issues. I close this.

closed 2020-06-15 by bastibl