bastibl/gr-ieee802-11 #228: using gr-lora module

roboticsai, 2020-07-08:
Is it possible to directly use the gr-lora module as transceiver with this block i.e gr-ieee802-11 without any modeification on the gr-ieee802-11 block? 
Or if some modifications are required for this block to work with gr-lora package, what kind of modefications is needed?

bastibl, 2020-07-08:
WLAN and LoRa are different technologies with no obvious similarities or standardized interoperability.

closed 2020-07-08 by bastibl

roboticsai, 2020-07-08:
Isn't it possible to convert the gr-lora block in such way that it communicates with gr-ieee block? By doing modification in gr-lora block/source code ( which is my plan ).

bastibl, 2020-07-08:
Again, WLAN and LoRa are different technologies. If you adapt LoRa to work with WLAN, you have to convert the LoRa transceiver to a WLAN transceiver. This doesn't make any sense.

roboticsai, 2020-07-08:
so instead of lora, will other radio transceivers like xbee work? For example first reading the data from radio and converting it to wlan suitable data and connecting that block with gr-ieee block. And is ip mesh imploemented/supported in this gr-ieee blcok for multiple compnuters mesh communication?

bastibl, 2020-07-08:
If you have a PC/node with a LoRa transceiver and a WLAN/ZigBee transceiver, you can, of course, receive data from LoRa and send it out again with WLAN/ZigBee. This is, however, unrelated to this module.

roboticsai, 2020-07-08:
ok. thanks for this information.