bastibl/gr-ieee802-11 #230: Error Raised When Running wifi_loopback.py

ChengDaHaI, 2020-07-22:
Hi,

My GNU Radio and gr-ieee802-11 version are both 3.8. My python version is 3.8.2. My OS is Ubuntu 20.04.

 I followed the instruction to build and install gr-ieee802-11. When i generate and run wifi_loopback.py file from wifi_loopback.grc, i got the following errors:

Error 1. 
```
self._snr_win = RangeWidget(self._snr_range, self.set_snr, 'snr', "counter_slider", float, QtCore.Qt.Horizontal)
TypeError: __init__() takes from 5 to 6 positional arguments but 7 were given
```
I solve the error by manually removing the latest parameter QtCore.Qt.Horizontal. Any idea about how to solve the problem forever?

Error 2.
```
File "/usr/local/lib/python3/dist-packages/ieee802_11/ieee802_11_swig.py", line 589, in make     
return _ieee802_11_swig.frame_equalizer_make(algo, freq, bw, log, debug) 
TypeError: in method 'frame_equalizer_make', argument 1 of type 'Equalizer'
```
It should be the error related to swig module (my swig version is 4.0.1.) Any idea about the error?

Thanks

bastibl, 2020-07-22:
You might have a cluttered setup with artifacts from different installations. It looks like the library that is used is `maint-3.8`, while the block definition of the RangeWidget is from a `master` installation. Make sure that there are no left-overs from previous attempts or a system-wide installation that is interfering.

ChengDaHaI, 2020-07-22:
> You might have a cluttered setup with artifacts from different installations. It looks like the library that is used is `maint-3.8`, while the block definition of the RangeWidget is from a `master` [...]

Thanks for the quick reply.

Yes, i have installed master first and then remove master and installed maint-3.8. I'll check if there are left-overs. However, does Error 2 still result from the left-overs of master? It seems to be related with swig module.

bastibl, 2020-07-23:
I'd suggests, you first fix your environment before looking into the other error. It seems rather strange and probably stems from a similar issue.

ChengDaHaI, 2020-07-28:
> I'd suggests, you first fix your environment before looking into the other error. It seems rather strange and probably stems from a similar issue.

Sorry for the late reply. I fixed the first error by editing /usr/share/gnuradio/grc/blocks/qtgui_range.block.yml:
changing the default  
`imports: from gnuradio.qtgui import Range, RangeWidget
`
to
```
imports: |-
    from gnuradio.qtgui import Range, RangeWidget
    from PyQt5 import QtCore
```
The second error is fixed by reinstallation.

closed 2020-07-28 by ChengDaHaI