bastibl/gr-ieee802-11 #231: Question on LO offset correction

mychele, 2020-07-23:
Hi Bastian,
I have a doubt on the  LO offset correction code in frame_equalizer_impl.cc, specifically regarding how the first two symbols (the LTFs) differ from the others. 
The LTFs have (if I am not wrong) [1, -1, 1, 1] as sequence in the "pilot" subcarriers, and you take this into account when computing beta for symbols 0 and 1. Similarly, d_er is computed and updated only after the LTFs, i.e., from symbol 2.

However, the current code uses the d_prev_pilots array, which for symbol 2 corresponds to the "pilots" for symbol 1, the LTF, which are stored with this code:
```
d_prev_pilots[0] = current_symbol[11] *  p;
d_prev_pilots[1] = current_symbol[25] *  p;
d_prev_pilots[2] = current_symbol[39] *  p;
d_prev_pilots[3] = current_symbol[53] * -p;
```
Following the logic for the beta computation, shouldn't this be:
```
d_prev_pilots[0] = current_symbol[11];
d_prev_pilots[1] = - current_symbol[25];
d_prev_pilots[2] = current_symbol[39];
d_prev_pilots[3] = current_symbol[53];
```
to store the pilots for symbol 1?
Or am I getting this wrong?
Thank you very much!
Michele

bastibl, 2020-07-23:
Hi Michele, good catch. I think, you're right. It should be set like you propose for symbol 0-and 1. Do you want to submit a PR or should I fix it?

mychele, 2020-07-23:
Hi Bastian, sure - will do. I see you have many branches - where should I
base the PR?

bastibl, 2020-07-29:
Sorry for the delay. I was not at the PC for a couple of days. You can base it on `master` or `maint-3.8` whatever is easier for you. I'll cherry-pick to the other branches.

mychele, 2020-07-29:
No worries, I opened #232

commit ae1286a7 (2020-07-30): Merge pull request #232 from mychele/maint-3.8

Fixes #231
closed 2020-07-30 by bastibl