bastibl/gr-ieee802-11 #235: Could you explain more on how to use wifi_transceiver.grc?

jamesshao8, 2020-08-18:
Hi,

Thanks for the project!

I successfully used wifi_rx.grc to demodulate commercial WiFi AP packet, using LimeSDR. And I am also able to transmit udp data by wifi_tx.grc to wifi_rx.grc. (wifi_tx only works on USRP B210).

I am now trying to use wifi_transceiver.grc.

I have two computers with 2 USRP B210. They are both running the wifi_transceiver.grc
Then I input the following command
On computer 1: sudo ifconfig tap0 192.168.200.1
On computer 2: sudo ifconfig tap0 192.168.200.2

After that, I can see commands output on the terminal. But when I try to ping Computer 2's IP address on Computer 1. There is no response, and vice versa.

I did try to play around with MAC address, like exchanging the SRC and DST, but it doesn't work.

I understand it's not normal wifi, but at least it should be able to ping each others, right?

bastibl, 2020-08-19:
This setup is a bit more involved, as it also requires manually configuring ARP. There is a nic.sh script in the apps folder that shows how to set it up. Also use Wireshark on both ends to debug the setup.

jamesshao8, 2020-08-19:
Thanks. I successfully established a virtual network now. I can ping or ssh from one computer to another.
Here's my simplified version of the script, in case someone else need it:
On computer 1:
terminal run:
sudo ifconfig tap0 down
sudo ifconfig tap0 hw ether 30:14:4a:e6:46:e4
sudo ifconfig tap0 mtu 440
sudo ifconfig tap0 up
sudo ifconfig tap0 192.168.123.2

sudo route del -net 192.168.123.0/24
sudo route add -net 192.168.123.0/24 mss 400 dev tap0

sudo tc qdisc add dev tap0 root netem delay 10ms

then run the wifi_transceiver.py
in wifi_transceiver.py, this line should be "self.ieee802_11_mac_0 = ieee802_11.mac(([0x30, 0x14, 0x4a, 0xe6, 0x46, 0xe4]), ([0x12, 0x34, 0x56, 0x78, 0x90, 0xab]), ([0x42, 0x42, 0x42, 0x42, 0x42, 0x42]))"

Then terminal run:
sudo arp -s 192.168.123.1 12:34:56:78:90:ab

On computer 2:
terminal run:
sudo ifconfig tap0 down
sudo ifconfig tap0 hw ether 12:34:56:78:90:ab
sudo ifconfig tap0 mtu 440
sudo ifconfig tap0 up
sudo ifconfig tap0 192.168.123.1

sudo route del -net 192.168.123.0/24
sudo route add -net 192.168.123.0/24 mss 400 dev tap0

sudo tc qdisc del dev tap0 root
sudo tc qdisc add dev tap0 root netem delay 10ms

then run wifi_transceiver.py
in which, make sure "self.ieee802_11_mac_0 = ieee802_11.mac(([0x12, 0x34, 0x56, 0x78, 0x90, 0xab]), ([0x30, 0x14, 0x4a, 0xe6, 0x46, 0xe4]), ([0x42, 0x42, 0x42, 0x42, 0x42, 0x42]))"

Then terminal run:
sudo arp -s 192.168.123.2 30:14:4a:e6:46:e4

closed 2020-08-19 by jamesshao8