bastibl/gr-ieee802-11 #236: send ssid beacon

jamesshao8, 2020-08-19:
When I use wifi_rx.py. I can decode the management packet, in which there is a beacon of SSID and its MAC address.
My question is if it's possibile to use wifi_tx.py to send a fake beacon with arbitrary SSID and MAC address?

Thanks.

bastibl, 2020-08-19:
This can be done using Scapy. Please see this presentation:
https://archive.fosdem.org/2019/schedule/event/gr_scapy/attachments/slides/3366/export/events/attachments/gr_scapy/slides/3366/gnuradio_meets_scapy.pdf

jamesshao8, 2020-08-20:
Thanks. I think I find what I need on page 21. Will give it a try.

jamesshao8, 2020-08-20:
Hi @bastibl ,
I downloaded scapy 2.4.3 and wrote the script according to PPT.
`from scapy.all import *
import socket
import time

frame = Dot11FCS(addr1='ff:ff:ff:ff:ff:ff', addr2='23:23:23:23:23:23', addr3 ='23:23:23:23:23:23')/Dot11Beacon()/Dot11Elt(ID='SSID', info='GR WLAN')
frame.show()

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
while True:
    sock.sendto(bytes(frame),("127.0.0.1",52001))
    time.sleep(0.1)`

And then start flowgraph wifi_tx.py to send the udp data to air, but my iphone can't found the ssid called "GR WLAN".

bastibl, 2020-08-20:
Make sure that you modified the TX flowgraph as shown in the presentation.

jamesshao8, 2020-08-20:
I am using wifi_tx.py, I checked the flow graph, and don't see much difference from the PPT.

I am now debugging with another SDR running wifi_rx.py
The packet looks like this:
=========================================
duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 3358
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
..........############..........d.....GR WLAN.T.D
LONG: frame start at 320
LONG: frame start at 149
encoding: 0 - length: 77 - symbols: 27
length: 73

new mac frame  (length 73)
=========================================
duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 3359
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
..........############..........d.....GR WLAN.T.D
LONG: frame start at 167
encoding: 0 - length: 528 - symbols: 177
length: 524

new mac frame  (length 524)

Which is a bit different from the normal ssid beacon sending from a commercial AP:
=========================================
duration: 00 00
frame control: 00 80 (MANAGEMENT)
Subtype: Beacon
SSID: "the ssid of AP"
seq nr: 
mac 1: ff:ff:ff:ff:ff:ff
mac 2: "mac of AP"
mac 3: "mac of AP"
new mac frame  (length 270)
=========================================

So, I suspect it's a problem caused by the packet I created.

bastibl, 2020-08-20:
The PDU should go directly to the PHY and not through he MAC (as shown in the presentation). Are you sure that this is the case for you?

jamesshao8, 2020-08-20:
I see your point. The PDU was going through MAC to PHY.

Now I disabled the MAC as follows.
[image]

But wifi_rx.py side can't decode packet now, the bpsk constellation looks normal though.

bastibl, 2020-08-20:
I guess you'll either have to debug this on your own or provide way more information about what exactly you are doing. I'm just blindly guess what you could have done wrong... Earlier you said that your iPhone doesn't pick it up, but here you send with a bandwidth of 5MHz, which is not supported etc etc.

jamesshao8, 2020-08-20:
Thanks. I will debug it by myself, and come back with some more results.

jamesshao8, 2020-09-21:
@bastibl Hi, I couldn't repeat the experiment with scapy. So I was trying another way.
I think what scapy does is to generate some data with SSID management packet, so I am trying to generate the same packet in the scope of gnuradio.

I re-enabled the MAC. And changed line 128 in /lib/mac.cc to "header.frame_control = 0x0080;"

My grc setup is changed as follows
[image]

And when I run the wifi_loopback.py, I can get something like this:
[image]

I think this is close to success, however when I do the same change to wifi_tx.py and transmitt the signal to the air, my phone ( including iPhone and Android) cant' find this SSID. I used frequency 2412MHz, bandwidth 20MHz, tried with BPSK QPSK QAM.

Please advice. Thank you.

jamesshao8, 2020-09-21:
By the way, This is the code I used with scapy, while MAC is disabled.
`from scapy.all import *
import socket
import time

frame = Dot11FCS(addr1='ff:ff:ff:ff:ff:ff', addr2='23:23:23:23:23:23', addr3 ='23:23:23:23:23:23')/Dot11Beacon()/Dot11Elt(ID='SSID', info='GR WLAN')
#frame = Dot11(addr1="ff:ff:ff:ff:ff:ff",addr2="23:23:23:23:23:23",addr3="23:23:23:23:23:23")/Dot11Beacon(cap="ESS", timestamp=1)/Dot11Elt(ID="SSID", info=RandString(RandNum(1,50)))/Dot11EltRates(rates=[130, 132, 11, 22])/Dot11Elt(ID="DSset", info="\x03")/Dot11Elt(ID="TIM", info="\x00\x01\x00\x00")
frame.show()

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
while True:
    sock.sendto(bytes(frame),("127.0.0.1",52003))
    time.sleep(0.1)`

The terminal shows:
`###[ 802.11-FCS ]### 
  subtype   = 8
  type      = Management
  proto     = 0
  FCfield   = 
  ID        = 0
  addr1     = ff:ff:ff:ff:ff:ff
  addr2     = 23:23:23:23:23:23
  addr3     = 23:23:23:23:23:23
  SC        = 0
  fcs       = None
###[ 802.11 Beacon ]### 
     timestamp = 0
     beacon_interval= 100
     cap       = 
###[ 802.11 Information Element ]### 
        ID        = SSID
        len       = None
        info      = 'GR WLAN'
`
This the flowgraph
[image]

But this method doesn't work for me either.
But it doesn't work either.

bastibl, 2020-09-21:
This doesn't look like a very systematic approach. You try your custom block in loopback (simulations) and Scapy over the air... There is much more to creating a beacon than changing the frame type. The parser in the Parse MAC block is very simplistic and not suited to debug frame formats. I would suggest to debug the Scapy approach and not try to reinvent it.

I'd suggest that you should first make sure that Scapy works in loopback (i.e., that frames show up correctly in Wireshark) and then try to send it over-the-air. The first should be trivial. If there are problems with the latter, I'd start with increasing the padding in the Packet Pad 2 block.

jamesshao8, 2020-09-22:
Thanks. I am now trying to send the packet created by scapy2.4.3 and through "WiFi PHY Hier", but I can't get any data out of it when analyzing with wireshark.

Later on, I did some research on how to use scapy to fake ssid with a normal wifi hardware in monitor mode, I succeeded sending my SSID beacon and discovered by another computer. However, when I feed the same frame to gnuradio, the result is still the same - nothing shows up in wireshark.

Here's my python code
`import socket
import time
from scapy.all import (Dot11, Dot11Beacon, Dot11Elt, RadioTap, sendp, hexdump)

SSID = 'GR WLAN'
iface = 'wlp7s0mon'
dot11 = Dot11(type=0, subtype=8, addr1='ff:ff:ff:ff:ff:ff', addr2='22:22:22:22:22:22', addr3='33:33:33:33:33:33')
beacon = Dot11Beacon()
essid = Dot11Elt(ID='SSID',info=SSID, len=len(SSID))
frame = RadioTap()/dot11/beacon/essid
frame.show()
hexdump(frame)
#sendp(frame, iface=iface, inter=0.100, loop=1)

while True:
    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    sock.sendto(bytes(frame),("127.0.0.1",52004))
    time.sleep(0.1)`

[image]

Sorry for bothering. I am trying to bring it into another open source gadget - portapack. I think it would be cool to send out a wifi beacon from a portable sdr device. But first I need to reproduce it in gnuradio with your program.

jamesshao8, 2020-09-23:
I found out that if I directly connect the Wireshark Connector block to Socket PDU (bypassing WiFi PHY Hier), I will get some output in wireshark. And if I use the frame same as your Slide (without radiotap), the frame looks alright in wireshark. I will see what has been changed by WiFi PHY Hier.

bastibl, 2020-09-24:
I'm not sure what you are doing. Now you changed the Dot11 header to a version without FCS... seems like you did some tests with WLAN card and did not switch back. I just tested it again with the loopback flowgraph and this script:

https://gist.github.com/bastibl/abedd2f8c656048152d3c30c249dc02e

It works without problems. Also you always paste the scripts wrong. I'm not sure if indentation is actually broken or if you just pasted it wrong.

jamesshao8, 2020-09-24:
Thanks. I will give it a try with your script. The indentation of my python script doesn't look right because it's changed when I try to post it here. It's alright when running in my computer.

pierregalant, 2021-01-21:
Hello, I made similar graphs, but I simply changed the USRP sink with an OSMOCOM block to work with the hackrf, but do not see the beacon over the air. Are these blocs compatible with the hackrf ?

I am able to receive beacon in loopback.

bastibl, 2021-01-21:
Yes, it works. Make sure that sample rate and frequency are set correctly for the osmocom sink. Also try different gains.

closed 2021-01-21 by bastibl

ShivaAlethea, 2024-01-25:
Hi @bastibl and @jamesshao8,
Working on the same using HackRF, But i'm not able to see any SSID / beacons on air.
- If i try to capture PCAPs directly from the **Socket PDU**, It is working fine and I'm able to see some **Beacons**.
- But the same is **not working** with **WiFi PHY Hier**,  No packets are captured.

PCAP With >> Socket PDU - **Working**
[image]

PCAP With >> WiFi PHY Hier - **Not working**
[image]

**If available can anyone please share me the ".grc" for this.**

bastibl, 2024-01-25:
The `mac_out` port only outputs the frames that are received (not the ones that are sent). So this is as expected.