bastibl/gr-ieee802-11 #238: Sorry for asking this again. wifi_phy_hier

nlhnt, 2020-10-19:
Hello,
first of all thank you so much for your work on the implementation of IEEE 802.11 TRx.
I have seen a lot of issues that were closed connected with "building" wifi_phy_hier.grc
I am looking for some directions.
I installed GNU Radio using my system package manager - I know this is not the best practice, but on the newest Fedora it was fairly up to date so I decided to give it a try.
I succesfully installed a custom module, following the guided tutorial on GNURadio wiki-site, it even appeared in my search bar in GRC.
Then I decided to dive a little deeper and try to install gr-foo, gr-ieee802_11, gr-ieee802_15_4 from source.
I cloned the repositories, made a build directory, ran the standard procedure:
```
$cd build
$cmake ..
$make // cmake --build .
$make test
$sudo make install
$sudo ldconfig
```
Anywho, the modules didn't give me any errors, I needed to install a missing dependency in my system (fftw library).
I got access to them in GRC, but the wifi_phy_hier.grc example gives me an error, which had been posted here a few time.
I didn't find the sollution though.
The module itself is imported correctly, so I don't know what's happeneing. It seems GRC also calls the same python interpreter.
I got some logs from calling the GRC through terminal [1] and from importing the gnuradio and ieee802_11 modules in iPython [2].

Basically the logs say that ieee802_11 module has no attributes `LS` or `signal_field`, which also seems to be the case when I import the module manuale in iPython [2].
Any idea what I might be missing or where I went wrong during the installation of the modules?
I feel like the closest issue which maybe has the sollution is #19 but  I am not reall sure if it is the same issue.
The only similarity seems to be that there the import was not succesful at all, here the module doesn't have attributes which are called for - which might have the same genesis?
As you can see in [2] the modules were built into the same directory ```/usr/lib64/python3.8/site-packages/```

[1]:
<details>
    <summary>Logs from GRC started from terminal - trying to load wifi_phy_hier.grc</summary>
<pre>
[komora@localhost gr_modules]$ gnuradio-companion
<<< Welcome to GNU Radio Companion 3.8.2.0 >>>

Block paths:
	/home/komora/.grc_gnuradio
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/komora/Dokumenty/gnuradio_projects/first_project/first_project.grc"
>>> Done

Loading: "/home/komora/Dokumenty/gr_modules/gr-ieee802-11/examples/wifi_phy_hier.grc"
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block chan_est
Traceback (most recent call last):
  File "/usr/lib64/python3.8/site-packages/gnuradio/grc/core/FlowGraph.py", line 257, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
AttributeError: module 'ieee802_11' has no attribute 'LS'
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block encoding
Traceback (most recent call last):
  File "/usr/lib64/python3.8/site-packages/gnuradio/grc/core/FlowGraph.py", line 257, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
AttributeError: module 'ieee802_11' has no attribute 'BPSK_1_2'
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block chan_est
Traceback (most recent call last):
  File "/usr/lib64/python3.8/site-packages/gnuradio/grc/core/FlowGraph.py", line 257, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
AttributeError: module 'ieee802_11' has no attribute 'LS'
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block encoding
Traceback (most recent call last):
  File "/usr/lib64/python3.8/site-packages/gnuradio/grc/core/FlowGraph.py", line 257, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
AttributeError: module 'ieee802_11' has no attribute 'BPSK_1_2'
>>> Done
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block chan_est
[... 289 lines cut ...]
module 'ieee802_11' has no attribute 'signal_field'
</pre>
<details>

[2]
<details>
<summary>Importing gnuradio and ieee802_11 modules in iPython</summary>
<pre>
[komora@localhost gr_modules]$ python
Python 3.8.6 (default, Sep 25 2020, 00:00:00) 
[GCC 10.2.1 20200723 (Red Hat 10.2.1-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gnuradio
>>> import ieee802_11
>>> ieee802_11
<module 'ieee802_11' from '/usr/lib64/python3.8/site-packages/ieee802_11/__init__.py'>
>>> gnuradio
<module 'gnuradio' from '/usr/lib64/python3.8/site-packages/gnuradio/__init__.py'>
>>> ieee802_11.
ieee802_11.encoding_to_parameters(  ieee802_11.payload_to_samples(      ieee802_11.unicode_literals
ieee802_11.mac_payload_to_payload(  ieee802_11.payload_to_symbols(      ieee802_11.utils
ieee802_11.payload_to_mac_payload(  ieee802_11.symbols_to_payload(      
>>> 
</pre>
</details>

nlhnt, 2020-10-19:
Closing for further investigation with the information found in #208

closed 2020-10-19 by nlhnt

nlhnt, 2020-10-19:
For anyone wondering, I strongly advocate using pybombs to install gnuradio and everything associated with it.
Even if cmake works then dependencies might not get resolved correctly inside the gnuradio-companion.
I tried installing.

This module with pybombs while having installed gnuradio through yum/dnf and it would get stuck on installing libvolk.
I tried installing libvolk from source instead and it also didn't work (even though it compiled and installed into system successfuly).

The solution is to use the same build configuration and dependency resolver which is smart: pybombs.
Install your gnuradio through pybombs, then add the modules using  the recipes and it will work.
It even installed libvolk correctly now...

dansharp23, 2021-06-11:
@nlhnt  I am trying to use pybombs and the gr-foo module worked for me but when I try to run 
`pybombs install gr-ieee802_11`
I get the error that there is no recipe file. Do you remember what recipe you had to add to pybombs for your install to work?
I added the recipe to pybombs via the command
`pybombs recipes add git://github.com/bastibl/gr-ieee802-11.git`
and it appears when I list the recipes.

bastibl, 2021-06-12:
I never used PyBomsb, but AFAIK, recipes are installation instruction for a module and not the module. So adding `git://github.com/bastibl/gr-ieee802-11.git` won't help. This module seems to be in `https://github.com/gnuradio/gr-recipes` and is called `gr-ieee80211`.