bastibl/gr-ieee802-11 #241: Error in WiFi Mapper block in examples/wifi_phy_hier.grc

AnnaSv, 2020-10-26:
I'm having an issue when trying to generate the wifi_phy_hier.grc flow graph provided in the examples folder. The Wifi Mapper block shows up red, and when I click on it to change the parameters, I see the following error:

```
Assertion "encoding in range(8)" failed.

Param - Encoding(encoding):
	Type "none" is not a possible type.
```

The 'encoding' variable is type int, not none. If I change the parameter to one of the options in the drop-down list, the error message stays the same. The flowgraph won't generate or run, and the only output in the GNU Radio Companion console is this:

```
 <<< Welcome to GNU Radio Companion 3.8.2.0 >>>

Block paths:
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/anna/gr-ieee802-11/examples/wifi_phy_hier.grc"
>>> Done
```

gnuradio and gr-ieee802-11 are built from source using the maint-3.8 branches, in Release mode. System is Ubuntu 18.04, Python 3.6.

bastibl, 2020-10-26:
I'm not able to reproduce this error. Any chance that there are left-overs from previous installation (attempts). Is this the first time you installed GNU Radio on this system?

AnnaSv, 2020-10-26:
Yes, there was! Ran "apt purge gnuradio", then reinstalled from source, and now it runs. Thanks!

berenguj, 2020-10-27:
Hi, I'm having a similar problem. The WiFi Mapper, Parameter, and WiFi Frame Equalizer blocks are all red due to missing definitions of the encoding, ieee802_11.BPSK_1_2, and chan_est variables. How would I fix this? Or are you able to give me the values of those variables so I can just replace them with those values? Thanks in advance!

[image]
[image]

bastibl, 2020-10-27:
Obviously you should also check if there are left-over artifacts from previous installations or installation attempts.

If your problem are actually the missing enums, this suggests that you compiled the module (`gr-ieee802-11`) without having Swig installed. In this case, your missing all C++ blocks, i.e. just replacing enums won't help. Please make sure that you have swig installed and rebuild the module. Make sure that cmake finds it during the `cmake ..` build step.

berenguj, 2020-10-29:
That worked! Thank you so much!

bastibl, 2020-10-29:
Great, thanks for the feedback. I'll close the issue.

closed 2020-10-29 by bastibl