bastibl/gr-ieee802-11 #242: basic data passing

berenguj, 2020-11-25:
Hi! I am very new to GNU Radio and SDR's and was looking for some advice/help. 

Since I'm still a beginner, the first thing I want to try to do is pass simple data, like an integer or a string of text, from one Ettus board to another (I currently have the two B210 boards, one can be the transmitter and one can be the receiver). I've downloaded your gr-ieee802-11 blocks and have ran the wifi_loopback.grc example and it seems like everything is working there. 

I've also looked at the wifi_tx.grc and wifi_rx.grc examples. From my understanding, in the wifi_tx.grc flowgraph, some sort of data from the Message Strobe block is passed through some intermediate blocks, then passed through the Wifi PHY Hier block, which I'm assuming acts as a transmitter. The Wifi PHY Hier block then transmits data to a UHD USRP Sink, which in my case would be my B210 Ettus board. My first question is what kind of data is the Message Strobe block producing? Would I be able to replace this Message Strobe block with a File Source to try to send data from a text file? My second question is, is my general understanding of how wifi_tx.grc is working accurate?

Also, from my understanding, in the wifi_rx.grc flowgraph, data is taken in from a UHD USRP Source, which in my case is the B210. Then that data is passed through some intermediate blocks in order to be able to decode the data I'm assuming. Then the received data finally goes through a Wifi Decode MAC and Wifi Parse MAC block which decodes the data I'm assuming. Then it can optionally go through a Wireshark Connector and File Sink, which I assume would allow me to write the received data to a new text file. Is my understanding correct?

Do you mind giving me some guidance on how to accomplish simple data communication between two Ettus boards? Again, I'm very new to this, so I appreciate any help you can give me!

bastibl, 2020-11-25:
The message strobe is just a demo that sends a string. This string is added after the WLAN header. Normally, there would be an IP header in real networks. You should be able to confirm that communication works by looking at the console output of the receiver or the PCAP file. (The output of the Wireshark Connector that is written to a file is not a text file as you assumed, but a PCAP file that can be read, for example, by Wireshark). The WiFi Parse MAC block will print some MAC header information and the payload. This is also what the loopback flowgraph does, which sends a given number of 'x's.
It might anyways make sense to experiment with different mechanisms with the loopback flowgraph.
- You can put a different message in the loopback flowgraph and see how the printed output changes (e.g., `pmt.intern("Hello World!")`)
- Instead of repeatedly sending the same message, you could add a Socket PDU block (configured as UDP server) and send data to the UDP socket. Any payload is then added after the WLAN header similar to the string.
There are many ways to send data to the UDP socket. For example a simple [Python script](https://github.com/bastibl/gr-ieee802-15-4/blob/maint-3.8/utils/udp.py) or use netcat `nc -u 127.0.0.1 52001`).

There are many more ways. If you have any specific needs, I can maybe give you more pointers.

armandsarkani, 2020-11-29:
Hey there, I am working on the same project as @berenguj and I was experimenting with trying to send a string from a GNU Radio flow graph via UDP to a simple server I have set up. I have created a message strobe and I am sending a PDU block which is a `pmt.cons` pair of a dictionary of key/values and an empty `u8vector`. I put this through "PDU to Tagged Stream", "Copy", and "Tagged Stream to PDU" blocks (not sure if any of those are necessary). Finally this PDU is sent via UDP to a server that is running in the background. However, since this data is being sent in this PMT PDU format, and not a string, I have no idea how to process this and convert it back into a string so my server can use this data. Right now when I send this PDU data to my UDP server, I just get an empty string. How can I convert this from this PDU format to a string on the server side? Thank you.

berenguj, 2020-11-29:
Thank you for responding @bastibl! I tested the loopback flowgraph using a different message like Hello World as you suggested and noticed the change in the output.
[image]
I also ran the transmitter on one B210 Ettus board and ran the receiver on another B210 Ettus board. I was expecting the output of the receiver to look similar to the output of the loopback flowgraph, but instead found it printing a bunch of 'O's. I'm not quite sure what this means. Am I supposed to be seeing a similar output as shown in the loopback flowgraph? If so, do you know what may be going wrong? If not, what do the repeated 'O's mean? (below is the output from the wifi_rx flowgraph)
[image]
I also just tried adding a Socket PDU block to the loopback flowgraph, disabled the message strobe block, and ran a Python script to see if I could send data through a UDP socket but it doesn't detect the message. My setup is shown below. Is there something wrong with my setup?
[image]
[image]
Thank you so much again for all of the help! I'm new to this so it is very much appreciated.

berenguj, 2020-12-01:
I realized that for my 3rd question regarding the loopback example using UDP, I didn't set the Socket PDU type to UDP Client. I set it to UDP client but still get no data from my python script that is sending messages to that UDP port.

bastibl, 2020-12-01:
@armandsarkani This is an issue tracker for the gr-ieee802-11 module. Your question, however, seems to be about GNU Radio in general. It should be posted to the GNU Radio mailing list. Anyways, I'm not sure what your flowgraph does, but you can just use Message Strobe -> Socket PDU (in this example I use UDP server mode). The message PMT should be set to `pmt.cons(pmt.make_dict(), pmt.make_u8vector(10, ord('x')))`, which sends 10 'x's. You can then connect to the flowgraph. For example with netcat (`nc -u localhost 52001`) (press <enter> once after the command).

@berenguj The idea of testing w/ loopback first, was exactly to rule out problems like configuring the socket wrong. The 'O's on the command line indicate that your PC doesn't process the data fast enough and, consequently, samples are dropped and frames are lost. This can have multiple issues. See here for some things you can try:
https://www.wime-project.net/installation/#overruns-and-underruns

If none work, try using a more recent PC.

armandsarkani, 2020-12-01:
Thanks @bastibl, I figured it was not a relevant question. I actually was just able to fix my issue with sending/receiving PDU data through UDP and it seems to work great using the ASCII to char method (`ord(x)`) and putting that in the `u8vector.`

berenguj, 2020-12-03:
Hi @bastibl I followed the suggestions to try to fix the issue of the dropped packets. I ran volk profile and I think it made a significant difference. Before running volk profile, I had received zero packets, but after running volk profile, I'm able to receive packets, but they're malformed. Additionally, the output in the GNU Radio terminal still shows the O's. I feel like the fact that the PCAP file has some sort of packets in it is a better sign. 
[image]

As for trying the other trouble shooting problems, I've messed around with the gain (increasing and decreasing the value) of both the rx and tx but it didn't seem to make a difference. I've also tried changing the period of the message strobe, changing the sample rate of the rx and tx, changing the lo_offset of the rx and tx, but have had no luck with getting the packet to send successfully.

How would I choose the correct value for the gain, lo_offset, etc? 

For reference, I have two B210 Ettus boards and am running wifi_rx.py and wifi_tx.py on a 2017 Macbook Air with macOS Catalina 10.15.3.

bastibl, 2020-12-03:
If there are still 'O's, your PC does not process the samples fast enough, i.e., there is no need to fine tune gain, etc, since there will always be frame losses... Just to be sure, you are running rx and tx on two different PCs?
Also since you are not interacting with a normal WLAN card, you can reduce the sample rate to 5MHz, which reduces the bandwidth but also the computational load. Make sure to tune to a channel/frequency where there is low interference from other devices. Otherwise the receiver might be busy with interference.
The malformed packets in Wireshark look bogus. Try the loopback flowgraph with a similar payload and look at the PCAP output. It should me the same.

berenguj, 2020-12-05:
Hi, I didn't realize that I needed to run rx and tx on two different PCs so I'm trying to do that right now.
I've installed GNU Radio and the gr-IEEE802_11 and gr-foo blocks onto another Mac. I'm trying to get wifi_rx.grc to run but am receiving this error for different python files:

```
Traceback (most recent call last):
  File "/usr/local/Cellar/gnuradio/3.8.2.0_5/lib/python3.9/site-packages/gnuradio/grc/gui/VariableEditor.py", line 197, in set_value
    evaluated = str(block.params[key].evaluate())
  File "/usr/local/Cellar/gnuradio/3.8.2.0_5/lib/python3.9/site-packages/gnuradio/grc/core/params/param.py", line 230, in evaluate
    raise Exception('Value "{}" cannot be evaluated:\n{}'.format(expr, e))
Exception: Value "[ieee802_11.LS, ieee802_11.LMS, ieee802_11.STA, ieee802_11.COMB]" cannot be evaluated:
name 'ieee802_11' is not defined
Traceback (most recent call last):
  File "/usr/local/Cellar/gnuradio/3.8.2.0_5/lib/python3.9/site-packages/gnuradio/grc/core/params/param.py", line 228, in evaluate
    value = self.parent_flowgraph.evaluate(expr)
  File "/usr/local/Cellar/gnuradio/3.8.2.0_5/lib/python3.9/site-packages/gnuradio/grc/core/FlowGraph.py", line 291, in evaluate
    return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace))
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
```
To troubleshoot, I've tried the suggestions in this issue https://github.com/bastibl/gr-ieee802-11/issues/6 but doing 'export PYTHONPATH=/opt/local/lib/python2.7/site-packages' didn't work. Is there anything else I can try? 

Again, thanks so much for all of the help it is so appreciated!

berenguj, 2020-12-05:
Hi, I didn't realize I needed to run rx and tx on two different computers, so I set up another Mac (2017 Macbook Pro with macOS Catalina 10.15.7) and ran the rx on that and the tx on my original Mac. I still get the same results as before with the O's printing. I also checked the Wireshark file. Before running volk_profile, there were no packets, and after running volk_profile the a malformed packet showed up. I also tried reducing the sample rate to 5MHz and changed the channel/frequency like you suggested but no luck.

Are there any other possible solutions? What kind of PC can process the samples fast enough?

bastibl, 2020-12-05:
It's not required to run them on different PCs, but it makes it much easier to set up and rules out many potential problems.
Your PC should easily be able to process the samples (especially if you reduce the bandwidth to 5MHz). I have no idea why this shouldn't work. Some ideas:
- make sure that SDRs are not directly side-by-side. Leave some space.
- add a FFT plot to see if the signal look as expected, i.e., are clearly visible at the receiver
- try different channels and gain settings
- make sure that there is no DC offset (the device supports DC offset correction)
- make sure you use the *unmodified* source and flowgraphs
Also, you switched directly from simulations to two SDRs. You should first test if you can receive frames from a WLAN card and send frames that are received by a WLAN card. At the moment it's not clear whether the sender or the receiver has problems. (For interacting with a normal WLAN card, you have to set the sample rate to 20MHz).

closed 2021-02-15 by bastibl