bastibl/gr-ieee802-11 #243: how to get packet more than just ack or beacon?

vincijoy96, 2020-12-01:
Hi! I am very new to GNU Radio and SDR's and was looking for some advice/help too, LOL.

since I am a Chinese, there may be some grammar or expression problems in my sentences, sorry for that.

I have tried to catch packets by hackrf one, recently, but I can just get ack or broadcast or something like that, I wanna catch ICMP or nomal packet, can you give me some help?

Looking for your reply, thanks in advance.

bastibl, 2020-12-01:
Other frames might be longer and/or sent with different modulation and coding schemes. This can make them harder to  decode, i.e., you'd require a higher SNR. So you can try to change parameters, make sure that no samples are dropped (overruns). Also since the HackRF has a DC offset, try filtering it. Here is a flowgraph (for an older version of the module) that comes with a DC filter. You can just copy the filter (which comes after the Osmocom Source) Note, the filter is not optimized and might introduce significant computational overhead.

https://github.com/bastibl/instant-gnuradio/blob/wlpc/assets/wifi_rx.grc

vincijoy96, 2020-12-01:
fine, I will try that, really thank you for your soon reply.

vincijoy96, 2020-12-01:
hi, 

I found that if I ping broadcast address then I can catch the ICMP packet, but if I ping other address then I can not, is there need to set something like promiscuous mode?: )

bastibl, 2020-12-01:
There is no promiscuous or monitor mode. You should see all frames by default. I'd recommend to use Wireshark on the host/interface that pings to better understand what's going wrong. I'd assume that, in your case, the frame is not sent out in the first place, since ARP resolution fails. I guess you want to ping an IP, your host tries to figure out the corresponding MAC address but fails. You can, however, configure ARP entries manually. See the `nic.sh` script in the `apps` folder as an example:
https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/apps/nic.sh#L41

vincijoy96, 2020-12-01:
I used command-line to pings a web site and got responses but wifi_rx.grc got nothing, but while pings broadcast it works, so weird, LOL.

bastibl, 2020-12-01:
Make sense, for the reason mentioned above...

closed 2020-12-01 by bastibl