bastibl/gr-ieee802-11 #245: adaptive filter least square

adjmxx, 2021-01-08:
Hi one question, the algorithm that you used in the channel equalizer is least square or RLS, because  only encounter information about LMS and RLS

bastibl, 2021-01-08:
There are several different equalizers. LS and LMS are least square. See here, for example: https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/equalizer/lms.cc#L38-L54

bastibl, 2021-01-08:
The module comes with four equalizers, including the LS and the LMS equalizer. Both calculate an initial estimate of the channel per subcarrier, using the long training sequence. This sequence repeats twice, i.e., you have two samples for each subcarrier. The best guess for the actual value is, therefore, just the mean of these two values. While people call this mean square, it such a simple case that there is no difference between mean, LS, or RLS.
https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/equalizer/lms.cc#L38-L39

adjmxx, 2021-01-10:
Thank you brother, was to me of so much help.

closed 2021-01-11 by bastibl