bastibl/gr-ieee802-11 #246: Running wifi_loopback on USRP device

armandsarkani, 2021-01-09:
Hello,

I was trying to modify the wifi_loopback example so that it runs on a single USRP B210 device, to see if I encounter any issues with over/underruns. Do you think this is something that is feasible, and if so, where should the USRP Source and Sink blocks be placed? 

Thanks

berenguj, 2021-01-10:
Hi, I'm working with @armandsarkani as well. Essentially, we're just trying to verify that our USRP B210 device can transmit and receive properly. First, we tried running these [examples](https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio) provided by Ettus to verify the operation of the USRP. When running the benchmark, receiving, and transmitting examples, we ran into overruns and underruns. We tried adjusting the tx and rx sample rates, but still ran into overruns and underruns. 

From there, we tried out this [FM Radio example](https://wiki.gnuradio.org/index.php/File:USRP_FM_fg.png) provided by GNURadio. This worked fine. Since it uses a USRP Source block, I assume that the board can probably receive just fine, but maybe it has trouble transmitting. I'm not sure though. 

Now, we're trying to see if we can modify the wifi_loopback example so that it can verify that our B210 device can transmit and receive properly, like Armand mentioned. 

Do you have any other suggestions on verifying that our B210 can successfully transmit and receive? Or any suggestions on how to eliminate overruns and underrruns?

Also, we have tested your wifi_tx, wifi_rx, and wifi_transceiver examples on two platforms, on a MacOS and on an Ubuntu OS on a Raspberry Pi. They both gave the same results, with overruns and underruns, so we've deducted that the problem is with our boards and not the environment the program is running on or the programs themselves.

bastibl, 2021-01-12:
I'm not sure how you're approaching this. Since the setup and benchmarks provided by the SDR vendor already exhibit this issue, you should ask there for help.
Also the experiment you're trying to setup (i.e., send and receive full duplex) is the most demanding in terms of CPU demand and bus speed. Once you made sure that your hardware is correctly setup, start with a simple configuration with two SDRs, one sending w/ 5Mhz and one receiving w/ 5MHz. If you just have one device, try to decode frames from a normal WLAN device (in this case you have to use 20Mhz).
If you use 20Mhz, you'll need a decent PC. A Raspberry PI will very likely not work out-of-the box. The fact that the PC that you tried was running Ubuntu doesn't tell much. Make sure it uses USB 3, a native installation (no VM), and a decent PC.

berenguj, 2021-01-13:
Okay, thanks for the advice!

closed 2021-02-15 by bastibl