bastibl/gr-ieee802-11 #248: IEEE 802.11 with Hackrf

pierregalant, 2021-01-18:
I am trying to connect your Wi-Fi transmitter to emit beacon using the hackrf but it does not work. I have simply put an osmocom block instead of the original one.

Does your Wi-Fi transmitter modulate the signal on a certain frequency. Or I can easily change the channel in the osmocom block.

How do you calculate the SNR in your receiver. Is the snr taken in reference to 1mW? Is it in db or dbm ?

Thanking you in advance.

bastibl, 2021-01-18:
The question the suggests you're rather new to SDRs. Maybe this is not a good first project and you want to have a look at GNU Radio tutorials: https://wiki.gnuradio.org/index.php/Tutorials
Some comments:
SDR applications (like the ones implemented using GNU Radio) usually produce complex baseband signals. The SDR sends them out on arbitrary frequencies. The Osmocom block has a frequency parameter. You should have configured the paramter to use the corresponding variable in the flowgraph. The GUI allows to adjust then WLAN channel (and hence the frequency) during runtime.

The SNR is a ratio. It is unitless.

pierregalant, 2021-01-18:
Hello bastian, yes I am new to SDR, I have a computer science background though I have a some knowledge in signal processing. I have seen the frequency parameter and though setting it to 2.472G or instance, I do not see the beacon. I have tried some tutorials and I am able to emit some noise and FM signals in a lab using the hackrf.

Yes you are right, SNR is a ratio, can you please tell me how the snr value parameter in the transmitter hierarchical bloc varies ? When I put 20 as value, I have values ranging from 18-24. How can these variations happen provided that you are multiplying the signal by a constant = 10**(snr/10.0))**.5.

bastibl, 2021-01-21:
What does it meant that you don't "see" the beacon.  And how did you look? A WLAN card? QT GUI? Fosphor? Dumped samples in a file an used Inspectrum. Depending on how you look, there may be many things that could have gone wrong. For example, was the WLAN card in monitor mode, did you adjust the sample rate correctly etc etc etc. These types of questions are really better suited for the GNU Radio Matrix chat (chat.gnruadio.org) or the mailing list.

Regarding SNR, what value varies? And why do you configure a SNR when you use the HackRF? You're question really don't make much sense to me. Please, if you have follow-up questions, put some more work in them... And if you have question regarding a calculation, then maybe tell me what you calculated to understand what doesn't make sense in your opinion...
http://www.catb.org/~esr/faqs/smart-questions.html

pierregalant, 2021-01-21:
Hello,  I am trying to make two types of experiments. I am emitting beacon via scapy in a loopback (100% simulation) and via scapy on air using the hackrf. For the simulation part, I am able to transmit and send receive the beacon. I have made a chain scapy->transmitter->receiver. I have modified your 802.11 receiver bloc to add wireshark blocs. For the simulation part, in wireshark, when I visualize the beacon, I have a SNR value in dbm which fluctuates +-4 dbm around the value I have inserted in your transmitter bloc as snr parameter. In your transmitter bloc, at some point you multiply the signal with a multipy constant bloc and you put this formula in the bloc: 10**(snr/10.0))**.5. I do not understand why you multiply the signal with this formula.

Concerning the hackrf, I have replaced the receiver bloc with an osmocom bloc to emit the beacon on air as the other person did with the USRP but I do not see the beacon on my phone when I perform a scan or with wireshark in monitor mode.

bastibl, 2021-01-22:
Thanks, this is way easier to answer :-)

Regarding SNR: the signal from the PHY block has an average power of 1. The noise is also fixed at 1. So if you want to have an SNR of 10, for example, you have to increase the power of the signal by 10. Since amplitude and power relate with A**2 = P, you have to take the square root of 10. Hence the **.5 at the end. the 10**(snr/10.0) is a conversion from dB (in which the SNR is specified) to linear domain.

Regarding the over-the-air experiment. Make sure that you also set the frequency, set the sample rate to 20MHz, try different gains, send frames at a high rate/as fast as possible (to increase the chance that the phone picks one up when it is scanning the channel). If you have a second SDR, check with fosphor or any other spectrum viewer if there is a signal where you expect it to be. (and if it has the correct handwidth, and if it has a good SNR).

pierregalant, 2021-02-03:
Hello, Ok I understand now why you have used this formula for the SNR.

However for the over-the-air experiment, I have modified your wifitx.grc to work with the hackRF (see below) but I still not receive the beacons on wireshark or on my phone. I have used RF lab equipment to see if the signal is where I expect it to be. The signal is on the desired frequency but my pc and my phone do not seem to be able to decode it. My pc and wireshark are properly configured to monitor mode and are able to see/decode other Wi-Fi beacons. I have also tried to replace the beacon by your message strobe and Wi-Fi MAC blocks but same problem. I can see it on the spectrum but my pc on wireshark does not see the data frames. 

Moreover, I have tried to connect two hackrf with an RF wire, same problem the receiver does not seem to be able to decode the packets.

 The chain only works when I have a chain UDP -> WIFITransmitter -> WIFIReceiver in a sort of  local loopback in gnuradio without air transmission. Under this configuration I can see both beacon and message strobe. This configuration is nonetheless not interesting for me. Could you please check my over-the-air hackrf Wi-Fi transmitter and receiver ? I have tried to reduce the beacon interval and message strobe period, but it does not help!

Thanking you in advance,
PL
Wi-Fi transmitter grc

[image]

Wi-Fi Receiver grc

[image]

bastibl, 2021-02-03:
I just pushed a commit to the `maint-3.8` branch that adds a Osmocom Sink. It is by default disabled, so you have to enable it first. I tested it with the HackRF and it works. My laptop can decode the frames. You seem to use 3.7, so you might have to update first.

Also, you did not just replace the USRP sink with a corresponding sink for the HackRF, you also deleted two blocks that were not only there for cosmetic reasons :-) I'd recommend to put the Multiply Const and Padding blocks back.

pierregalant, 2021-02-03:
Hello, 

I tried with your wifi_tx.grc which has the multipy const and padding blocks at first, but I was not receiving the beacon neither. And yes good point, I forgot to add them in this chain.

Anyway, I will try to update to 3.8 tomorrow. Thanks for your help. Apart from your thesis, have you published a paper which explains the logic behind each bloc in details ?

pierregalant, 2021-02-04:
It works on 3.8 thanks !

closed 2021-02-04 by pierregalant

bastibl, 2021-02-04:
Great, glad it worked!