bastibl/gr-ieee802-11 #250: PlutoSDR With 802.11 Module

Bricko44, 2021-01-19:
Hello,

Having succeeded in making GNURadio work correctly with my environment and the 802.11 module, I want to start implementing my tests.

I want to set up an RX-TX link using 2 Adalm-Pluto SDRs. Then I want to place a spectrum analyzer (external) in the middle of the link to run some mesures. For that, I saw that a TX program, and an RX program are present in the Example folder. I have tried to implement Tx with a PlutoSDR Source rather than a USR Source, but I cannot get the whole thing to work (In fact, I don't see any packets transiting, but I think i'm not doing the right thing in only replacing one block).

I'm still not very comfortable with GR, and i don't know if it's possible to send packets from a PlutoSDR, and receive them on another PlutoSDR (on the same computer ? or a second computer ? one using wifi_rx.grc, and the other wifi_tx.grc) ?

Thanks in advance for your help and you advices,
Hugo

bastibl, 2021-01-21:
What do you mean with "seeing packets transiting" how did you look? When replacing the source/sinks make sure to set the sample rate and center frequency as well. Also, maybe you want to start with a smaller sample rate (5MHz) (I'm not sure if or how well the Pluto does for 20MHz). Also the gain for the USRP might have been normalized (0--1). Make sure to adjust the gain for the Pluto (It might have no normalized gain.)
The simplest setup is using two Plutos with RX/TX flowgraphs on separate PCs.

Bricko44, 2021-01-21:
Hello Bastibl, and thanks for your reply.
 
After several tests, we have succeeded with my colleague, in setting up a connection with two PCs, and one PlutoSDR per PC.
We replaced the USRP blocks with PlutoSDR blocks, and as you just mentioned, we re-parameterized what was needed. The link is working.

After that we started sending broadcast messages while the link was established. These messages are visible on the GUI of the received signal. In addition we also connected a 1.7-2.7GHz antenna to a spectral analyzer to see our signal.

Regarding "seeing packets transiting", we have therefore managed to see the packets sent. However, we would like to enable wireshark to view our packets. Unfortunately we are unable to get wireshark to work. This is launched on the Rx computer. It does create the wifi.pcap file, but once opened it is still empty.

bastibl, 2021-01-21:
... still don't get how message are visible in the GUI. You see the transmission in the spectrum? You see the autocorrelation of the preamble? You see some constellations that somehow look correct? You see the actual decoded frame in the terminal (not the GUI)?

In other words, are you actually sure that the message is decoded (in which case, it would be strange if the Wireshark file was empty) or is there just some signal that looks good?

Bricko44, 2021-01-21:
On the spectrum analyzer, we can clearly see our signal with the chosen bandwidth (in our case, 4MHz).

Regarding the sending of packets, there is indeed a problem, since the packets do not leave with the same format as the wifi_loopback program. We don't understand why, since we haven't touched anything on the wifi_tx program apart from the PlutoSDR block. We therefore do not see our decoded frame on the RX side.

For the constellations, on sending, it looks correct, but on receiving there is indeed something very abstract.

bastibl, 2021-01-21:
Please Hugo.... I have no idea what you mean with frames don't leave it the same format..... Also you didn't touch wifi_tx except for the Pluto. But then there are no constellations. There is no plot. I have no idea what you did or what you're looking at. This is really frustrating.... On the receiver you see "something very abstract".........

closed 2021-01-21 by bastibl