bastibl/gr-ieee802-11 #251: Make error during gr-ieee802-11 installation

mjihad-75, 2021-02-02:
Hello,

I'm installing gr-ieee802-11, and during the installation process I found a failure during the compilation using make due to conflicts with the "boost::shared" library, although I managed to solve the failure by replacing all the matches of this library with "std::shared", but now I get the following error:

[image]

I am using gnuradio version v3.8.0.0-943-g17c7fccf as seen in the following image:

[image]

any solution for this error? Thank you very much

bastibl, 2021-02-02:
Looks like you have multiple versions of GNU Radio installed. While the `gnuradio-config-info` is from 3.8. The headers/libraries you're linking against seem to be from GNU Radio 3.9.

mjihad-75, 2021-02-02:
thanks! what is the way to solve it? I've tried removing gnuradio and installing it but it still persists.

bastibl, 2021-02-02:
Please... wouldn't it be possible to spend 30sec and formulate a question that could, at least, be answered? This is frustrating...
I don't know your system. I don't know what GNU Radio versions you tried to install and how. I don't know which branch of this module you tried to install. I don't know what you did when you "tried removing gnuradio".

While there are some ways to do it right, there's an infinite number of ways to do it wrong. So since you don't provide any information apart from the fact that you tried *something* that was not working, I wonder what you expect from me. A book on "Every possible mistake you can make when installing software on Linux -- including solutions for every distribution"?

http://www.catb.org/~esr/faqs/smart-questions.html

You can try this: Delete `/usr/local`. Then, if your distribution comes with GNU Radio 3.8 in its package manager, install it from there (including the -dev packages and swig). Then recompile the maint-3.8 branch of gr-ieee802-11. If your distribution doesn't come with 3.8, just install it from source. But 3.8 and not again 3.9.

closed 2021-02-15 by bastibl