bastibl/gr-ieee802-11 #253: Using USRP B200 mini and GNU Radio

Raggen96, 2021-02-15:
Hi,
I am new to GNU Radio and USRP. I am trying to use a USRP B200 mini to monitor Wi-Fi packets with GNU Radio and I have downloaded your project. Should this work without making any changes to the flowchart? I have tried running wifi_rx.grc and in the generated .pcap file, there are only two packets and both says “ Malformed Packet: 802.11 Radiotap”.

bastibl, 2021-02-15:
With a B200, you don't have to change the flowgraph. During runtime you can set the sample rate (set to 20MHz to interact with a normal WLAN card), the channel, and the gain (try different values). This should work out-of-the-box. I'd recommend to try on the 5GHz band. If you have to use 2.4GHz for whatever reason, make sure that it is 11g with no compatibility modes.
The frame might be malformed on the higher layers in Wireshark, but it's a valid WLAN frame with MAC etc.
If you have problems getting it to work, please provide more information.

Raggen96, 2021-02-16:
Hi,
Thanks! This worked. However, now I am getting this error. Have you received the same?
[image]

bastibl, 2021-02-16:
No, I've never seen this. Did you get this once or do you get it all the time?

Raggen96, 2021-02-17:
Ok, I got this several times, but not all the time. Will try to do some adjustments.

Raggen96, 2021-03-01:
Hi again. 

This works fine now thanks! I have an additional question. I am running the wifi_tranciever.grc file in gnuradio-companion. Are packets captured in monitor mode? 

Regards

bastibl, 2021-03-02:
Since you seemed to have figured out the problem, it would be nice if you could explain it here or open a pull request.

Regarding monitor mode: this is not a WLAN device but just a PHY. It doesn't have monitor or promisc mode. This would be implemented on higher layers. The receiver will log all frames.

Raggen96, 2021-03-03:
Hi, 
I never figured out the problem. The message in the picture above was only displayed on channel 44 (5GHz). I realized that the packets I was looking for, Wi-Fi Aware (NAN) packets, were sent on channel 149 and channel 6. Pcap files from these channels did not get this error message. 

Thanks for the explanation on monitor mode!

closed 2021-03-25 by bastibl

MountainLogic, 2021-03-25:
Just as a note for others, if there is sufficient noise to corrupt bits in the packet and produce an invalid checksum in a recorded pcap file then WireShark should tag that packet as corrupt (as it should).   - Not sure if Bastian's code passes up corrupt packets.
A common mistake with WiFi/Bluetooth and SDR is to try and plug a consumer WiFi antenna into a SDR.  Most removable consumer antennas have reverse polarity SMA connectors while professional equipment like most SDRs have normal polarity on their SMA connectors and will not pass the signal (look at the gender of the center pins of the connectors).  With the wrong polarity you can will either damage the connectors or not pass a signal with either case resulting in a very poor signal and a high rate of corrupt packets (or no packets).

bastibl, 2021-03-28:
Frames that do not pass the CRC check are dropped and won't show up in PCAP.