bastibl/gr-ieee802-11 #256: abnormal peaks

ENB213-xiaoh, 2021-02-23:
Hi Bastian. I'm trying to use your tool to measure the carrier frequency offset. Before that, I tried to reproduce the experiments shown in your paper <An IEEE 802.11a/g/p OFDM Receiver for GNU Radio>. But I get sort of weird results. I'm using N210.

For example, this image detects the frame start by correlation. The peak=1 are the beacon frames and it makes sense since there are about 2M samples between each two as the beacon interval is 102ms. However, there are some other peaks, and the rest values are all large, which is abnormal. The CP detection and LTF detection are showing the same thing. I'm sure there are no frames except beacons. So, I'm wondering why these abnormal peaks and high corr-values exist. Please kindly advise. Data are collected using sink block.

[image]

bastibl, 2021-02-23:
This seems to be an uncompensated DC offset. The USRP source should have an option to filter it out.

ENB213-xiaoh, 2021-02-24:
Yes. Thanks. I have cleaned out the DC offset. I have to update my gnuradio because it seems the old version is not able to enable it automatically.

closed 2021-02-24 by ENB213-xiaoh