bastibl/gr-ieee802-11 #257: Based on the receiver, how to get separate the packet from the received data?

YunxiangCai, 2021-02-25:
Dear Dr. Bastian Bloessl,

I am doing a project about online vehicular direction finding based on a Deep Learning Method. The basic idea is to classify the angel of arrival (AOA) according to phase difference  of received signals from different antennas. I use the USRP N210 to receive the signal and GNU radio to perform signal process. I need to separate the received packet from the whole received data and your 802.11 receiver is a wonderful tool to realize this.

My Problem
[1] refers that the the process of the receiver can be divided into two stage: Frame Detection and Frame Decoding. After frame Detection, the packet can be separated from the received data. But when I add a file sink after the OFDM Sync Long moudle, I found the separated data is not exactly the packet. So how could I separate the packet(including signal filed and data field) from the received data?

[1] Bloessl B, Segata M, Sommer C, et al. An IEEE 802.11 a/g/p OFDM Receiver for GNU Radio[C]//Proceedings of the second workshop on Software radio implementation forum. 2013: 9-16.

YunxiangCai, 2021-02-25:
[image]

I add a file sink before FFT. But seems that the signal is not the pure packet data.

bastibl, 2021-02-25:
I'm not sure what exactly you were expecting. You sent the signal over the air. It is of course not a 1-to-1 equivalent of the IQ data that you produced.
In case you wonder about the length: The frame detection just recognizes a frame start. It doesn't know where it ends. So it just forwards data according to a max sized frame into the flowgraph.

YunxiangCai, 2021-02-25:
Thanks for your reply. Sorry for my poor English. I add a diagram to explain it:

[image]

The received data from USRP is massive but not all data contains information. When there is no packet, there is no information, so I want to extract the data that contains a packet.

**WIFI Sync Short** moudle and **WIFI Sync Long** moudle play a role to detect a packet and forward the packet data to Frame decoding. I wonder if it can be done with the receiver.  

I have tried to add a file sink before FFT:
[image]

bastibl, 2021-02-25:
As I said: The frame detection just recognizes a frame start. It doesn't know where it ends. So it just forwards data corresponding to a max sized frame into the flowgraph. Your frame is shorter, which is why you see these gaps.
The normal flowgraph, will learn form the signal field, how long the frame is an drop the rest. If you want to remove the gaps, you'd have to add this functionality yourself. Depending on your goal/scenarios, it might be enough to just add a threshold.

YunxiangCai, 2021-02-25:
Thanks a lot. It works.

closed 2021-03-25 by bastibl