bastibl/gr-ieee802-11 #258: Why is there an offset in OFDM Sync Short block?

YunxiangCai, 2021-02-25:
> We implemented the OFDM sync short block to act like a valve. Its inputs are the samples from the USRP and the normalized autocorrelation coefficient. If it detects a plateau in the autocorrelation stream, it pipes a fixed number of samples into the rest of the signal processing pipeline; otherwise it drops the samples. [1]

Why is there an offset in the COPY stage (as shown in the red box in the below screen capture
)? Why not just use **out[o] = in[o]**? And what's the meaning of the offset?

[image]

[1] Bloessl B, Segata M, Sommer C, et al. An IEEE 802.11 a/g/p OFDM Receiver for GNU Radio[C]//Proceedings of the second workshop on Software radio implementation forum. 2013: 9-16.

ENB213-xiaoh, 2021-02-26:
I think this should be the frequency offset correction. The frequency offset is only measured from STF in this work. In the paper, he wrote this part into Sync Long block. But I bet he wrote the code here as the correlation results are already available in sync short block, which is in_abs[]. You may double check that there are no codes calculating FO in sync long.

bastibl, 2021-03-01:
Yes, this is the frequency offset correction. There is a first frequency offset estimation and correction in Sync Short and another more fine grained one in Sync Long.

closed 2021-03-25 by bastibl