bastibl/gr-ieee802-11 #262: IEEE 802.11 n/ac transceiver

elisebsm, 2021-03-25:
Hi,

Is it possible to modify this to be able to monitor packets sent over IEEE 802.11 n/ac?

Thanks!

bastibl, 2021-03-25:
This is possible but requires significant work. Also it is unlikely that the high-bandwidth and MIMO modes will run in real-time.

julled, 2021-03-29:
But there are also other interesting things like STBC and LDPC in the n standard, which would be very interesting especially for long range links. 
Would this be possible to implement on consumer hardware ? Or maybe on lower datarates?

bastibl, 2021-04-13:
LDPC would work. STBC might be a bit more tricky to get it to work in real-time. These features are also part of 11ah, which uses more narrow channels. This would be an interesting standard to implement (which very likely works in real-time). Based on this, one could try the 11n/ac features.

closed 2021-05-25 by bastibl