bastibl/gr-ieee802-11 #263: CSI for AoA estimation

hsujim2, 2021-03-27:
Hi Bastian,
I have install gr-ieee802-11, I copied all block in wifi_rx's for three times to get three antenna's CSI. I had read another issues and log the csi from d_H in equalizer, however the CSI phase of the center frequency is almostly zero. So three antenna didn't have differential phase and can't use for AoA estimation. Am I log the wrong things or ignore something?
Also, the phase didn't stable and change rapidly even in 5Ghz, which has less interference.
Thanks!

bastibl, 2021-03-28:
What equalizer are you using? I don't see why it should be 0. Also, what do you mean with center frequency? The carrier in the middle of the spectrum is not used (DC subcarrier). Anyways, the three receivers operate independently, i.e., do frequency offset correction and synchronization independently. You'd have to annotate the values they estimate and use for correction to do AoA estimation downstream.

hsujim2, 2021-03-28:
Thanks for your reply.
Sorry, is my fault, actually isn't the middle subcarrier, is closed the middle. 
This is for three antenna CSI phase, 
[image]
and this is differential phase while Subtract to antenna 1
[image]
I try all equalizer, the result look similar.

I know three antenna have different offset, but it only cause AoA estimation not acurate, but now AoA estimation cannot work, while I change the angle of antenna, the angle it estimate won't change.

bastibl, 2021-03-28:
Can you try disabling these corrections in the modified frame equalizer:

https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/frame_equalizer_impl.cc#L151-L153
and
https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/frame_equalizer_impl.cc#L194-L196

hsujim2, 2021-03-29:
After disabling, phase changed!
But it still unstable.
I have filter it in decode_mac, only log the csi from one MAC address, it still unstable.
I record it and post on Youtube.
https://www.youtube.com/watch?v=XA53Lx8b_y0
I have another question, the MAC address in decode_mac is origin device's MAC or destination?
If is destination device's MAC, then I did it wrong.

hsujim2, 2021-04-01:
I'm sorry for asking question again.
I trying to testing the stability of csi phase, so I use wifi_tx to send packet.
However it didn't work, I read another issue and change packet pad, but still didn't work.
The LED of tx isn't on and nothing read from wifi_rx. I use GUI sink to monitor the output signal, it did have signal but usrp didn't send it.
I try other examples and it works, so USRP is working.
My device are usrp X310 with CBX-120 and UBX-160.
Thank for your help.

bastibl, 2021-04-13:
The decode MAC block doesn't look at MAC addresses. Parse MAC prints 3 MAC addresses, the second one corresponds to the transmitter. You can use Wireshark to check this.
A WLAN frame is very short and you just might not see the LED turning on. Use a WLAN card and try to receive frames or another SDR with gr-fosphor to see if there are transmissions. It's easier to spot if you increase the rate of frames that are sent.

hsujim2, 2021-04-19:
Thanks for your help. In the end, I found that I can't disable corrections is frame equalizer, it will cause wifi signal can't be decode.
I just read the multiply of them.
I'm sure that X310 cannot send wifi signal, because rx didn't receive anything, while using N210 and B210 it works fine.
And now I had these alternative so it won't bother me.
But now I have face two new problem
1. I try to sync three rx for CSI AoA estimation, so I connect the threshold together(cor in sync short) to ensure three stream have same origin. It success and work quite well in two antenna in one X310, but when it comes to two X310 with three antenna, then signal from second X310 cannot decode wifi signal, constellation is in a mess. I had connect the PPS and ref in to octaclock to sync two machine, do you have any suggestion for sync the signal? or where am I did wrong cause these problem.
[image]

2. I try to use wifi rx on B210, but while I switch to 20Mhz mode, then the app stop working, while 10Mhz work fine. I guess it because computer cannot handle it. My CPU is i7-8700, the performance it quite enough. I have no idea why these happen.

bastibl, 2021-04-19:
I'm not sure, I understand your questions. But the problem is harder than just duplicating the receive chain. It will require more logic to keep everything in sync. Apart from timing, it might also happen that one RX chain detects and/or synchronizes on a frame, while the other doesn't. Your receive needs additional logic to deal with this case. If you have more questions, please provide a bit more details.

hsujim2, 2021-04-20:
I know MIMO can't just use these simple logic, but now I use cable to test it, so no phase different in two channel. Both rx chain recieve frame almost at the same time.
I feel wired why these can work
[image]
But these can't
[image]
I have considering timestamp and using delay try to find the delay between two USRP, but it make no effort.
I have use sine wave and are sure they are in same frequency and sync successfully.

bastibl, 2021-04-20:
Sorry, but I don't get what you are talking about. If you still have issues, please describe them more clearly.

hsujim2, 2021-04-20:
Difference is in the left of the images.
There are two streams connect share the same sync signal.
The first picture is while connected to one usrp machine, which has two daughterboard
The second picture is while connected to two different machine with each single daughterboard
The window on the right is the constellation of two streams.
If I can't do this simple way to sync two streams, then the first picture's scenario won't work, however it works.
But when it comes to dual usrp, then it can't work.
I had connect PPS and Ref clock to same source to sync two machine.

bastibl, 2021-04-20:
My understanding is that you have two separate receive chains that do not interact. That means it cannot be a synchronization issue. Also, since the CPU load should be similar in both cases, I'd assume that there are no overruns in any cases. I'd assume it is a networking, driver, or HW issue, e.g., two devices compete for bandwidth on an Ethernet link or two instances of the driver. As far as I see, it is not an issue of this module.

hsujim2, 2021-05-20:
Thank for your response before, you really help me a lot. I solve several problem but now I face a new problem.
The phase difference between RF's isn't stable even I connect four RF daughterboard to single antenna using three T junction connector.
In Sbx, the phase will have two different slope, it's weird but I can solve it by referencing two ubx's CSI phase different and calibrate it. 
Both SBX and UBX CSI phase difference will jump unexpectedly, and these effect my AoA estimation.
Like these picture
[image]
Most phase difference are intensive, but sometimes it jumps.
I try to turn off 
https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/frame_equalizer_impl.cc#L151-L153
and
https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/frame_equalizer_impl.cc#L194-L196
that you had told me before, but it become even worse, so I just multiple them for complete CSI output.
I'm sure the CSI is from same frame(using frame control field wifi seq number from module) and I read the second symbol in frame equalizer, I had try read the CSI from first symbol, nothing change.

My devices are:
Rx0,Rx1:UBX-160 on  X310
Rx2,Rx3:SBX-40 on  X310
two x310 use octoclock to sync PPS&clock
Using wifi card to send ping packet to AP and USRPs are in monitor mode.

bastibl, 2021-05-25:
If your oscillators are not in sync, you'll just see random phase offsets. So I'm not sure what you are trying to do. If or how this can be achieved with your devices is something you'd have to ask Ettus or post to the USRP-users mailing list.

YunxiangCai, 2021-06-23:
> Thank for your response before, it really help me a lot. I solve several problem but now I face a new problem.
> The phase difference between RF's isn't stable even I connect four RF daughterboard to single antenna using three T junction connector.
> I expect to get stable phase different cause they had same channel information, and move them to zero in order to calibrate but:
> 1.In Sbx, the phase will have two different slope, it's weird but I can solve it by referencing two ubx's CSI phase different and calibrate it.(green and blue line)
> 2.Both SBX and UBX(red line) CSI phase difference will jump unexpectedly, and these effect my AoA estimation.
> Like these picture
> [image]
> Most phase difference are intensive, but sometimes it jumps.
> I try to turn off
> https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/frame_equalizer_impl.cc#L151-L153
> and
> https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/frame_equalizer_impl.cc#L194-L196
> that you had told me before, but it become even worse, so I just multiple them for complete CSI output.
> I'm sure the CSI is from same frame and same TX antenna(using frame control field wifi seq number and address2 from module) and the CSI I read is from the second symbol in frame equalizer, I had try read the CSI from first symbol, nothing change.
> 
> My devices are:
> Rx0,Rx1:UBX-160 on X310
> Rx2,Rx3:SBX-40 on X310
> two x310 use octoclock to sync PPS&clock
> Using wifi card to send ping packet to AP and USRPs are in monitor mode.
> 
> By the way, I notice that the rx LED on USRP turn off sometimes, I don't know if it had some connection between them

We hava met a similar problem. Could I have your email and discuss about this problem?

hsujim2, 2021-06-26:
> We hava met a similar problem. Could I have your email and discuss about this problem?

ok

closed 2021-07-13 by bastibl