bastibl/gr-ieee802-11 #265: about wifi_transceiver.grc

zsy9512, 2021-04-14:
hello！ I try your example for wifi_transceiver.grc with a pair of B210 devices.I run the program successfully as your instructions in the nic.sh, however, each device can only recieve the signals  from himself. I can't make a connection between the two devices. I don't know how to do with them,can you give me same advices?Thank you!
Best Wishes!

zsy9512, 2021-04-14:
I tried to just transmit at device A, and have received the signals at device B clearly .So why I cann't get the signals at device B when I run the wifi_transceiver at A?

bastibl, 2021-04-15:
This is probably a networking issue (misconfigured ARP entries, IPs, routing, etc). Start Wireshark at sender and receiver and see where things go wrong.

zsy9512, 2021-04-20:
> This is probably a networking issue (misconfigured ARP entries, IPs, routing, etc). Start Wireshark at sender and receiver and see where things go wrong.

Thanks! I checked my configuration and passed the test.But I meets a new issues when I test the awgn model.It shows "thread[thread-per-block[4]: <block mac (25)>]: pmt_uniform_vector_elements: wrong_type : ()" in the terminal.
Howerver,I can not locate the mistake!

bastibl, 2021-04-20:
It looks like an edge-case where you send or receive an empty frame.

zsy9512, 2021-04-21:
I find it caused  by "random_periodic_msg_source_impl.cc" and "wireshark_connector_impl.cc" files in gr-foo.When the source stop the  singal generation send "EOF" tag, wireshark cann't exit at the same time

bastibl, 2021-04-21:
Great that you figured it out. I tried to reproduce but failed, but I'm also not fully understanding the issue, since the MAC block seemed to fail. If you could create a PR that would be very appreciated.

closed 2021-04-21 by bastibl