bastibl/gr-ieee802-11 #266: Short Frames

stephen3303, 2021-04-15:
Hello,
      I am studying the 802.11 protocol and your module has been very helpful in my learning. I am working on a project to locate access points by MAC address and signal strength using a yagi antenna.  This type of capability already exists using a traditional adapter and airodump-ng or kismet. My goal is to implement on a SDR to examine 802.11 protocols (BPSK, QAM) on non-standard frequencies. 

Setup:
Bare-metal machine
Intel i7, 32Gb RAM
Ubuntu 20.04
GnuRadio 3.8.2.0
UHD 3.15.0.0-62-g7a3f1516
Ettus B210

Observations:
Currently, I am capturing 802.11g/20Mhz traffic at my residence and noticed many short frame messages in the output among correctly decoded frames. I have used the wifi_rx and wifi_transeiver examples with success. Adjusting the rx_gain does not seem to effect the number of short frames messages received. I have noticed the amount of short vs decoded goes above 50% on busy channels. From my understanding of the protocol and the flowgraph the messages indicate that a frame start was detected; however, when attempting to decode there are not enough bytes to generate a frame header. My observations have been that these short frames are always either 16 or 10 bytes long.

Why do I care:
I am curious if this is an indication of needing to adjust variables to tune the frame detect for my noisy environment. The repetition of size 10 and 16 leads me to be suspicious, but I am not sure of the cause. Running the flowgraph with mac parser debug on typically result in it continuing to output in the console for a long time after I have closed the GUI. Maybe reducing the number of the short frame messages I get will lessen the load on the console. I working to add some GUI components to display the decoded MAC addresses with signal strength to avoid the heavy traffic on the console output. Adding functionality to calculate noise level would allow for a SNR values.

See example output below.

new mac frame  (length 16)
=========================================
frame too short to parse (<20)

new mac frame  (length 10)
=========================================
frame too short to parse (<20)

Should I be concerned with the >50% of frames being short frames in the console. Does this indicate loss packets?
Do you have any thoughts as to why it is always 10 or 16 vs random?

Thanks, 

Stephen

bastibl, 2021-04-15:
thanks for the great detailed description of your issue. This output comes from the Parse MAC block, which is just a very simple basic debug block. And it mainly means I didn't implement a parser for this frame. What you see, might be RTS/CTS frames. These frames are not dropped, just not parsed for console output. They will show up in Wireshark. You can cross-check the console output with your Wireshark trace. Maybe I should just delete this message from the Parse block.

stephen3303, 2021-04-15:
Thank you for the quick reply. I am looking at wireshark and the smallest frame lengths are Acknowledgement and CTS at 27 bytes and CF-End at 33 bytes. 

I am matching sequence numbers of beacon frames between console and wireshark. Wireshark shows them to be length 298 and the console indicates that they are 281.

It looks like the frame length between the console and wireshark do not match with wireshark being 17 more then console. 

If this is all accurate then that correctly identifies the size 10 and 16 short frames as the acknowledgement and CTS.

bastibl, 2021-04-16:
No, that's fine. Wireshark adds metadata to the frame (the pcap, radiotap headers). They include things like time, modulation, snr, etc. These additional headers make up the difference. The actual data of the WLAN part should be 10/16 bytes.