bastibl/gr-ieee802-11 #267: #wifi_transceiver.grc

Joeyue68, 2021-04-21:
Hey!
I used  USRP B210 as device. But when I attempted to generate wifi_transceiver.grc, the red light on the TXRX port, which was on the device, didn't glint. It seemed that I could receive the signal, but I couldn't transmit signal into air from B210. I wonder where the bug is. And whether there is some problem on my device or .grc. Plus, I could trasmit signal when I generate the wifi_tx.grc. This  truly bothers me. I'd appreciate it if you happen to know how to fix it!

zsy9512, 2021-04-21:
You can try to add your tx_gain

Joeyue68, 2021-04-21:
> You can try to add your tx_gain

Could you be more specific? Should I add a block in my wifi_transceiver.grc?

bastibl, 2021-04-21:
The frames are so short that you probably won't see the LED turning on. What exactly is your issue and setup? How do you know that it doesn't work etc.

Joeyue68, 2021-04-21:
/home/berk/桌面/1.png
/home/berk/桌面/2.png
These might be the problems, do you know how to fix them?

Joeyue68, 2021-04-21:
[image]
[image]
These seem to be the issues, could you help with that?

bastibl, 2021-04-23:
That's an issue of the GNU Radio QT GUI Sink and not of this module. Please write to the GNU Radio mailing list, ask in the Matrix chat, or open an issue at the GNU Radio repo.

closed 2021-04-23 by bastibl