bastibl/gr-ieee802-11 #269: How to transmit and receive messages using different frequency simultaneously?

egoistzty, 2021-04-22:
Hello! I am trying to use a usrp B210 to transmit and receive messages simultaneously, with different frequency. 

One of the board transmits message in one sub-channel, for example 5.87MHz, using RF A:Tx/Rx. And at the same time receive message in another sub-channel, 5.92MHz, using RF B:Tx/Rx.And the other board transmit in 5.92MHz and receive 5.87MHz;

I simply combined the two flowgraph (WiFi_rx,WiFi_tx) together into one gragh, and give the corresponding parameters above,but unfortunately neither of the boards can receive the correct messages from the other (actually even no correct constellation can be received)

So is there anything wrong with my operation, or the B210 just cannot transceiver in different channel at the same time? If it can,would you tell me how to realize it? 
Thank you very much.

bastibl, 2021-04-23:
Before testing this more complex setup, did you actually make sure that you can receive something in the simplest setup (i.e., wifi_tx -> wifi_rx)?

egoistzty, 2021-04-23:
Yes，actually I have tested 2*2 mimo before (with two boards transmit and receive using the same frequency)
and they work properly，and some collisions happened at times，so I want to use different freqs to avoid collisions.

bastibl, 2021-04-23:
Hmm,... I didn't test this setup. But since the other configuration works for you, it's a matter of using/configuring UHD. Maybe you could ask on the UHD mailing list.

egoistzty, 2021-04-23:
Okay，thank you for your reply (^_^)

closed 2021-05-25 by bastibl