bastibl/gr-ieee802-11 #271: The wifi_rx.grc can't demodulate 802.11g signal.

qw-Zhang, 2021-04-27:
Hi bastibl,
The wifi_rx.grc cannot demodulate 802.11g signal, i.e. setting the Frequency at 2412M in wifi_tx.grc and wifi_rx.grc. The constellation pionts are scatterring in graph and no information is generated in ".pcap" file.
However, the Frequency is setted at 5.89G(default), the wifi_rx.grc and wifi_tx.grc are work well. By the way, I use the usrp-2922, which is support 400M-4.4GHz. I don't know what excatly Crarrier Frequency is used (maybe up to 4.4GHz?). 

I guess the alogrithm that need to change something in wifi_rx.grc... For now, I failed. The output is repeat as screenshot
[image]

OS: ubuntu 18.04
Gnuradio: 3.8.2
gr-ieee802.11: maint-3.8
Hardware: Two USRP-2922

Thanks!

bastibl, 2021-04-27:
There is nothing changed in the receiver when switching the channel. I don't have this particular SDR, but maybe it is
- interference on 2.4GHz
- the received signal strength is different on 2.4GHz (antenna, frontend) -> try changing RX/TX gain
- there is a DC offset at 2.4GHz (try enabling DC offset compensation)
- try a different channel on the 2.4GHz band (to understand if it's the whole band or this particular channel)

qw-Zhang, 2021-04-28:
Thanks!
The reason is the RX&TX gain. Now, the wifi_rx.grc & wifi_tx.grc work well. 
However, the wifi_rx.grc cannot stop when I push "kill the flow graph"; the command still output and process about five minutes. I think that the computer processing ability maybe not enough power. Anyway, the constellation and command output the correct results at least.

Thanks again!

bastibl, 2021-04-28:
great, thanks for the feedback.

closed 2021-04-28 by bastibl