bastibl/gr-ieee802-11 #272: problem of wifi_rx.grc

lms7657061, 2021-05-19:
hello, my English is poor, so please ignore my grammar problems. I have two questions
1. Why I open the captured data packets with Wireshark, some of which are not 802.11 protocol, and why I can't  receive the beacon frame or probe frame sent by the mobile phone in the 2.4G frequency band?

lms7657061, 2021-05-26:
well, thank you very much! I have correctly received and demodulated the frame.

lms7657061, 2021-05-26:
> Can you please add much, much more information. For example, what frames you see and what exactly you're doing to receive beacons on 2.4GHz? Also, I'd suggest starting with 11a on 5GHz to rule out potential issues with unsupported frames on 2.4GHz (11b and compatibility modes).

well, thank you very much! I have correctly received and demodulated the frame.

lms7657061, 2021-05-26:
by the way, in the sync_short.cc file, whether the d_freq_offset is the coarse carrier frequency offset of the frame? in the sync_long.cc file ,whether the d_freq_offset is the fine carrier frequency offset of the frame? and how to print them?
sorry, I can't put the code picture in here.

bastibl, 2021-05-26:
great that it worked. The estimation in sync short is not as precise as the one in sync long. So you might call it coarse and fine. You can print variables with `std::cout << var << std::endl;`.

closed 2021-05-26 by bastibl