bastibl/gr-ieee802-11 #275: Ask question of the carrier frequency offset of the frame

lms7657061, 2021-05-28:
hello, sorry to bother you again!  In the sync_long.cc, there are d_freq_offset, in the frame_equalizer_impl.cc, there are d_freq_offset_from_synclong, but they have different values, why? and which one is the carrier frequency offset?
[image]
[image]

bastibl, 2021-05-28:
Sync short estimates the frequency offset, compensates for it, and annotates the estimate to tell downstream blocks how much it compensated. Sync long does another compensation based on more data and annotates the overall frequency offset (compensation of sync short + sync long), which is extracted in the equalizer. I'd assume `d_freq_offset_from_synclong` is what you are looking for.

lms7657061, 2021-05-28:
Ok，I get it, thank you very much!

closed 2021-05-28 by bastibl