bastibl/gr-ieee802-11 #277: Periodic corrupted OFDM sympols in wifi receiver

ChengDaHaI, 2021-06-03:
Hi,

I am using gr-ieee802-11 (main-3.7) to collect wifi datasets on USRP X310 with two CBX-120 Dboards (with subdev:A serving as TX and subdev:B serving as RX).  

However, when I look into the BER (after Viterbi decoding), I found something wields, as shown in the figure.
[image]
In this BER vs bit index plot, the BER is averaged over 3000 wifi packets (QAM16, 10MHz, 2.467GHz). Also, I found that each high peak (16 in total) corresponds to the (96 ) bits demodulated&decoded from one OFDM symbol. 

Motivated by this, I enumerate the packets with high errors in those high peak bits and then plot the following two figures 
[image]
[image]

Note that the eqSYM is the data in equalized OFDM symbols. 

I analyze the decoded bit from the mac_out port of WIFI_PHY_Hier (with my modified WIFI_Decode_MAC to output all wifi packets, ignore the CRC results) as well as the raw IQ output of USRP receiver followed by MATLAB implemented WIFI receiver. Both give me the save BER plot. 

In addition, I have tried the wifi_loopback.grc and find there is no such high peak in BER curve. Also, I have tried the same setting one another testbed where there is UBX-160 Dboard in USRP X310 and only 3 peaks appear

[image]

Thus, I guess the problem occurred in the USRP hardware side. The hardware related setting are:

-  lo_offset = 0MHz
-  uhd.tune_request(freq, rf_freq = freq - lo_offset, rf_freq_policy=uhd.tune_request.POLICY_MANUAL)
-  Mboard clock rate 200MHz
- TX normalized gain 0.7, RX normalized gain 0.1
- The distance between TX and RX antenna is about 1.5 meter
- The Dborad calibration is finished.

I would very appreciate if you could give some comments on how to debug the issue.

Best,
Hai

bastibl, 2021-06-04:
Thanks for the great, detailed description. Is it correct that the MATLAB implementation exhibits a similar BER pattern, i.e., it's not a bug in the code? (It might yet be a common shortcomming of both implementations to deal with the hardware impariment.)

The setup with the other X310, is that also RX and TX on one device? If so, can you try to split RX and TX. There might be oscillator leakage.

Did you try offset tuning?

Are there warnings/errors/overruns/underruns indicated?

ChengDaHaI, 2021-06-04:
> Thanks for the great, detailed description. Is it correct that the MATLAB implementation exhibits a similar BER pattern, i.e., it's not a bug in the code? (It might yet be a common shortcomming of both implementations to deal with the hardware impariment.)

- In the MATLAB implementation, I got the same BER plot (as shown). So it is not a bug in the code.

> The setup with the other X310, is that also RX and TX on one device? If so, can you try to split RX and TX. There might be oscillator leakage.

- Yes, in the two examples, TX and RX are both on one device. When you say split RX and TX, do you mean using two X310 with one serving as TX and another serving RX?

> Did you try offset tuning?

- DO you mean the lo offset tuning? If so, yes, but the default one in your example : "uhd.tune_request(freq, rf_freq = freq - lo_offset, rf_freq_policy=uhd.tune_request.POLICY_MANUAL)".

> Are there warnings/errors/overruns/underruns indicated?

- Yes, there are one warning related with URSP: "[WARNING] [MULTI_USRP] Setting IQ imbalance compensation is not possible on this device". Some underruns happens.

bastibl, 2021-06-07:
My guess was that there is some leakage between RX and TX. What do you mean, when you said you tried LO offset tuning? The command you posted is by default issued with an lo_offset of 0. So no offset tuning. It is a parameter that can be configured through the GUI and for offset tuning should be set to a value that is larger than bandwidth/2. If you already tried this, I'm running a bit out of ideas. Maybe some other interaction between RX and TX on the same device. I'd recommend to use two devices, one acting as RX and one acting ax TX. At least, it'll help to better understand the problem.

bastibl, 2021-08-28:
Wow, thanks for the great report. I don't have a really good idea what could be the problem. Just some ideas:
- If possible, you could have a look at the time domain signal and see what's going on. I'm not sure if this is an issue of the signal (as received by the SDR) or some issue with the receiver.
- If it is an issue with the receiver, the first thing that comes to mind is the `d_epsilon0` and `d_er` in the frame equalizer. Maybe you could set this to 0 for testing (also commenting out their updates). https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/frame_equalizer_impl.cc#L147-L148
- You could log the estimated frequency offset (it's annotated in a tag) and check if the values of the corrupted frames are different from the good frames.

closed 2022-04-11 by bastibl