bastibl/gr-ieee802-11 #278: No module named 'gnuradio.ctrlport'

ramziB99, 2021-06-08:
Hello, 
I have an issue when I tried to run wifi loopback example, im on Mac OS 11.4 with gnu radio 3.8.3.0.
Traceback (most recent call last):
  File "/Users/ramzib/gr-foo/build/gr-ieee802-11/examples/wifi_loopback.py", line 40, in <module>
    from gnuradio.ctrlport.monitor import *
ModuleNotFoundError: No module named 'gnuradio.ctrlport'

I did some research about that and installed thrift 0.14, unistalled then install again gnu radio but I got the same error. 

best, 
Ramzi

bastibl, 2021-06-09:
You can just disable/delete the Control Port blocks in the flowgraph. They are not required.

What exactly happened is hard to tell. If you reinstalled from the package manager, it will reinstall with the same feature set, independent from Thrift. If you compiled from source, you'd have to check the cmake output to understand why control port is not enabled (e.g.,  missing Thrift Python bindings). But that's more of a question for GNU Radio anyway.

ramziB99, 2021-06-10:
hi, 
Thank you, I disabled the ctrl port block and it worked. 
I will close the issue.

closed 2021-06-10 by ramziB99